stereolabs/zed-ros-wrapper #447: ZED wrapper TF tree

ThanasisTs, 2019-07-10:
Hello,
in commit `5d0d7fe1fda347d49f54faf23001b5376b1e8f19` i see that the tf tree did not contain the "base_link" or the "odom" frame. However, in the current version of the wrapper it does. Is there any particular reason for that change? If not, would it be possible to rename the "base_link" name with another one (e.g. "camera_link") since "base_link" is a very common name for a tr tree? Thanks in advance.

Myzhar, 2019-07-10:
We added the base_link to the default TF tree because it is easier to insert the camera into an existing robot configuration. If your robot URDF has a base_link you must only change the ZED URDF providing the right transform from zed_camera_center to base_link.
In this way the ZED wrapper takes care of the transformation of the pose calculated by the tracking module referring it directly to the robot position and you must not write your own node to calculate it.

If your configuration does not work with base_link you can change the ZED URDF according to your requirements remembering to change the `base_frame` parameter according to your reference frame

ThanasisTs, 2019-07-10:
Since almost any robot URDF contains a "base_link" frame, I don't quite understand why the insertion of this frame in the ZED URDF makes it easier to insert the camera into the existing robot.

Myzhar, 2019-07-10:
Because the ZED generates `odom` and `pose` in map frame, and we must refer them to the robot position, not to the camera position, otherwise users must create complicated nodes to convert the ZED topics to the correct frame

ThanasisTs, 2019-07-10:
I think I understand what you mean, but I believe it is still not necessary to connect those specific names of frames on the base camera link. Since when using any type of camera with a mobile platform, the odometry message is going to be generated by another node (e.g. robot_localization), the `odom` and `map` frames will introduce more problems, since they are far too common names that are generated by localization procedures.

For example, I tried renaming the `base_link` frame to `zed_camera_link` (in all the required files), and I could successfully access the camera's pose on the odom frame in rViz. Hence, the `odom` and `map` frames could be renamed to `camera_odom` and `camera_world`, and it would not create any problems with any robot localization method, since the extra non-standard frames would be connected to the tree, with the (robot) base_link --> zed_camera_link transformation. To be more precise, the `camera_odom` topic (not frame), that would be published from the zed wrapper, could be the input of a robot localization procedure, that combined with other inputs (IMU, encoders) would generate the final `odom` message.

Myzhar, 2019-07-10:
We provide all the parameters to perform the operations that you are suggesting nad we chose to expose those parameters according to our experience and according to users experience.

What you are suggesting is an advanced use of the ZED camera wrapper and the parameters have been added just for advanced users like you that require a different configuration from the default that we provide.
The "launch" files provided with the wrapper are more like an example about how to use it then files to be used for a final robotic project.

Just an example, for a robotic project that I'm developing I do not use the original "launch" file of the ZED Wrapper, but I used them as a base to write new launch file with the configuration required by the robot.

Myzhar, 2019-07-10:
*To rename the `base_link` simply set the parameter `base_frame` in [common.yaml](https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/params/common.yaml#L20)
*To rename the `odom` topic simply change the parameter `odometry_topic` in [common.yaml](https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/params/common.yaml#L61)
*To rename the `odom` frame name simply change the parameter `odometry_frame` in [common.yaml](https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/params/common.yaml#L52)
*To rename the `map` frame name simply change the parameter `map_frame` in [common.yaml](https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/params/common.yaml#L62)

ThanasisTs, 2019-07-12:
Ok thanks for the explanation. I was just trying to find a way to avoid these conflicts.

closed 2019-07-12 by ThanasisTs

Myzhar, 2019-07-12:
@ThanasisTs if you need an exhaustive guide for the ZED ROS Wrapper you can find it here:
https://www.stereolabs.com/docs/ros/zed_node/
All the parameters are explained and you can use the node in advanced mode

willzoe, 2019-12-27:
> If your robot URDF has a base_link you must only change the ZED URDF providing the right transform from zed_camera_center to .

How to set up a transformation from `zed_camera_center` to the `base_link` of an existing robot in a launch file?
Or, do I have to modify the source code in `zed.urdf`?

Myzhar, 2019-12-27:
@willzoe yes, you must modify the urdf.
What I usually do when I configure a robot is:
1) copy the originale ZED urdf file in my robot folder
2) copy the zed launch file in my robot folder
3) modify the launch file copy to point to the position of the urdf file copy
4) modify the urdf file copy to match the configuration of my robot

In this way when you update the ZED wrapper the launch file and the urdf file are not modified by the update process.

willzoe, 2019-12-27:
Thank you. 
By the way, if my robot has a `robot_state_publisher` node that publishes robot urdf, do I need to publish the zed urdf using the same node? Or directly use the `robot_state_publisher` node in `zed_camera.launch`, such as `zed_state_publisher`?

Myzhar, 2019-12-27:
You can use another `robot_state_publisher`, but be sure that all the nodes have a different name, otherwise the last launched will close the previous one.

willzoe, 2019-12-27:
What happens if two non-same `robot_state_publisher` nodes each publish a link named `base_link`?

According to my observation, the `base_link` released by the latter node will overwrite the former node published in the TF tree.

[image]
This is the TF tree of turtlebot, which contains the transformation from `base_footprint` to `base_link` broadcast by `/robot_state_publisher`.
[image]
This is the zed TF tree, which contains the transformation from `odom` to `base_link` broadcast by `/zed/zed_node`.

But when I start turtlebot and zed at the same time, only the transformation from `odom` to `base_link` broadcasted by `/zed/zed_node` is left in the TF tree, but the transformation from `base_footprint` to `base_link` broadcast by `/robot_state_publisher` is not visible.
[image]

Myzhar, 2020-01-10:
It's correct because the ZED node publishes the updated transform from `base_link` -> `odom`, while the turtlebot publishes `base_link` -> `base_footprint` -> `odom` and this is not compatible as you can notice running the command `roswtf`.
You can take two ways:
1) remove `base_footprint` in the Turtlebot (I think it is not useful)
2) add `base_footprint` to ZED URDF

willzoe, 2020-01-11:
Thanks.
Now I let `zed_node` no longer publish `odom-> base_link` TF and` map-> odom` TF, and use RGB-D odometry in `rtabmap` to publish `odom-> base_footprint` TF.

[image]

But it seems that `odom-> base_footprint` TF has some problems. I think it should be published by `rtabmap` instead of `mobile_base_nodelet_manager`.

willzoe, 2020-01-13:
I closed the `odom-> base_footprint` TF published by `mobile_base_nodelet_manager` via modifying the parameters of turtlebot ’s base `kobuki`.
[image]

Myzhar, 2020-01-13:
That's what's you should do.