stereolabs/zed-ros-wrapper #448: can not catkin_make the zed-ros-wrpper

guan4286, 2019-07-15:
when I run catkin_make -DCATKIN_WHITELIST_PACKAGES=zed-ros-wrapper;I get nothing.
[image]

how to solve it ?

Myzhar, 2019-07-15:
Hi @guan4286 
you are using the right command in the wrong way.

The `CATKIN_WHITELIST_PACKAGES` requires the name of the package to compile, but you are using the name of the repository.

In the github repository there are different package names:
- `zed_wrapper` -> The ZED ROS wrapper
- `zed_display_rviz` -> Rviz tools
- `zed_depth_sub_tutorial` -> depth subscribing tutorial
- `zed_tracking_sub_tutorial` -> positional tracking subscribing tutorial
- `zed_video_sub_tutorial` -> image subscribing tutorial
- `zed_ar_track_alvar_example` -> Example about how to use Alvar tags
- `zed_nodelet_example` -> Example about how to use the nodelet
- `zed_rtabmap_example` -> Example about how to use RTABmap

guan4286, 2019-07-15:
[image]
when I use catkin_make ,I get something like the above picture.

guan4286, 2019-07-15:
The official tutorial is to run "catkin_make"
[image]
what's going on and how to understand it ,then solve it .
Thank you very much

Myzhar, 2019-07-15:
In your issue you wrote that you were running the command
`catkin_make -DCATKIN_WHITELIST_PACKAGES=zed-ros-wrapper`
that's not a simple `catkin_make`.

If you run `catkin_make` without parameters, as you read in the official tutorial, you can compile all the packages without any problem.

Myzhar, 2019-07-15:
Can you post the result of the command 
`$ ls ~/catkin_ws/src`

guan4286, 2019-07-15:
[image]
Still have that problem.

Myzhar, 2019-07-15:
Yes, that was not a "solution" command, but a way to understand what's happening. It's like if you catkin warkspace is corrupted.
Can you try to delete `build` and `devel` folders in your catkin_ws folder and to compile again the whole workspace?
```
$ rm -rf ~/catkin_ws/devel
$ rm -rf ~/catkin_ws/build
$ cd ~/catkin_ws
$ catkin_make
```

guan4286, 2019-07-15:
Still have no Executable file under the zed_wrapper package.
[image]
[image]

Myzhar, 2019-07-15:
Ok, we are near to solve the issue. There is a configuration problem for one of the packages, but the error is not visible in you screenshoot.
Can you search for it running again the `catkin_make` command?

guan4286, 2019-07-15:
This is a official tutorial on ros wiki 
[image]

Myzhar, 2019-07-15:
That's not updated, however it should work. 
Please try to follow the official Stereolabs documentation:
https://www.stereolabs.com/docs/ros/

guan4286, 2019-07-15:
how to solve the configuration problem for one of the packages?

Myzhar, 2019-07-15:
If you don't tell me the exact error you are getting, I cannot find the solution.
Please copy and paste here the full output (no screenshoot) of the `catkin_make` command, so I can get what's wrong

guan4286, 2019-07-15:
I am loading it again,wait for a moment .
And I create a new workspace ,I am going t o try it later.

guan4286, 2019-07-15:
In my new workspce ,there is no any wrong . I really don't know why.
[image]
It works well

Myzhar, 2019-07-15:
Maybe that there was a problem with one of the other repositories that you have in the other workspace that stopped the compilation process.
If you need to use the other packages you can try to import them one by one to new workspace to search for the one that is missing dependencies.

guan4286, 2019-07-15:
Thank you for your  patience !

Myzhar, 2019-07-15:
You are welcome

closed 2019-07-15 by Myzhar