stereolabs/zed-ros-wrapper #449: Drift in ZED-M

shakeebbb, 2019-07-17:
I have connected my ZED mini camera with Jetson TX2 and walked in a loop multiple times. I am visualizing the data in RVIZ. I cannot get rid of the drift. It seems like it's also not performing loop closures. The spatial memory is 'true' in the launch file and I've tried increasing and decreasing the frame rate and the resolution as well. Am I missing something especially for the loop closure part? 

[image]

Myzhar, 2019-07-18:
Hi @shakeebbb 
can I ask you about what version of the ZED SDK are you using?

shakeebbb, 2019-07-18:
I'm using the latest setup available i.e., ZED SDK v 2.8.2 with Jetpack 4.2.

Myzhar, 2019-07-18:
It seems a problem not strictly related to ROS, but more generally related to the SDK. 

Can I ask you to write an email to `support@stereolabs.com` so we can open a ticket about it?
If you can please record an SVO in the same conditions of the test you made and attach it to the email (better if you can provide a link to download it) and the `conf` file of your camera that you can find in the folder `/usr/local/zed/settings/`.

Thank you
Walter

clungzta, 2019-09-24:
Was there any resolution for this?

I seem to be experiencing the same issue. It appears that no loop closures are being detected (despite spatial_memory=true). I am experiencing significant drift. Thus, the mapping is near unusable.

It results in poor quality maps from ZED ROS Wrapper (map=true), and RTAB-Map.

I am using TX2 with ZEDM, 2.8.3 ZED SDK, Jetpack 4.2.1.

See images:
[image]
[image]
[image]

(Note: I have the ZED head-mounted, and am seated rotating slowly and smoothly in my office chair to create those maps.)

obraun-sl, 2019-09-25:
You can try with TrackingParameters::enable_imu_fusion at false to check if the loop closing is performing better.

shakeebbb, 2019-09-26:
@clungzta I needed a quick solution for a lab project, I postponed the idea of using ZED odometry till the issue is resolved. Please let me know if ZED loop closure works out for you.

clungzta, 2019-09-26:
Thanks. However, setting enable_imu_fusion=false had no discernible improvement. Following the earlier suggestion, I'll record an SVO and my conf file, and raise a support ticket.

Myzhar, 2020-03-23:
Positional tracking has been improved with latest SDK versions.
I close the issue

closed 2020-03-23 by Myzhar