stereolabs/zed-ros-wrapper #450: ZED mini camera depth information fluctuation

bluespirit2, 2019-07-19:
Hello there,
Recently, I am engaged in ZED camera based object detection where i can get depth information about the detected objects. The depth information fluctuates in the range of 1 meter difference and even more sometimes. I have checked this by holding the camera in front of a wall. May i have your help on how to fix this fluctuation problem? Thanks in advance.

Myzhar, 2019-07-19:
Hi @bluespirit2 

can you publish a screenshoot of the wall that you are using as target?
What is the distance between the camera and the wall?

Thank you
Walter

bluespirit2, 2019-07-19:
Hi Myzhar,
Thank you so much for your prompted reply.
I focused the camera on a plain white wall located 3m from the camera. I measured the distance to the wall using meter for verification. The camera reads sometimes 2m and sometimes about 5m.

Myzhar, 2019-07-19:
The white wall is not a good target for a pure stereo vision sensors.
Stereo vision algorithms requires visual features to be matched on the two images acquired by the two camera sensors to correctly calculate the depth.

Even if the depth module of the ZED SDK is really advanced and it can estimate measures when a few information is leaking, it cannot be precise when the information is totally missing (that's the case of a surface with a homogeneous color).

That's why you are getting a very variable measure.

bluespirit2, 2019-07-20:
Thank you so much Walter. I will try on a wall with heterogeneous color and report you the output.

bluespirit2, 2019-07-22:
Hi Walter,
I just tested the depth information. Your advice is so fruitful that there is no fluctuation problem when the camera focuses on object with heterogeneous color. Thank you very much.

Myzhar, 2019-07-22:
You are welcome. I close the issue

closed 2019-07-22 by Myzhar

takahiko-hasegawa-hdjp, 2024-03-14:
Hi there, 

We have a similar issue with the ZED camera (but with ZED 2i camera). If there is an object in front of the camera (<1.6meters) the point cloud value becomes Infinity. We do understand that the accuracy of the depth/point cloud degrades as the object becomes further from the camera but did not expect it to be immeasurable/inaccurate when the object is close to the camera specifically less than about 1.6 meters.

We are using Ultra mode so should be accurate enough but not sure why this happens. The problem is that the issue occurs intermittently but when it does happen it's mostly in the beginning after the camera starts running (we are using ros2 so when the camera node start running). So we cant debug the issue very well. 

Does the Zed camera starts with some sort of auto-calibration? do we need to wait for some time after the camera node finished initialising?

Please let us know what could be done to resolve this issue.

Detail:
- custom ROS2 camera node using Zed (v4) sdk
- Ultra mode
- C++ 
- Inital parameter is mostly set to default 

Thanks in advance!