stereolabs/zed-ros-wrapper #451: Big latency in ZED M camera location publish rate

WondesenB, 2019-07-19:
Hello,
I am using ZED mini Camera for location estimation and object detection application. When I subscribe to location in map frame only the publish rate looks fine. However, when I subscribe to both location in map frame and image for detection  the location publish rate slow down more than by half and this is below the requirement for my application. I would appreciate if anyone shared me a solution for this. Thank you !

WondesenB, 2019-07-19:
Thank you very much for the reply! I am using Jetson TX2, and subscribing to rostopics "/zed/zed_node/pose" and "/zed/zed_node/rgb/image_rect_color".

Myzhar, 2019-07-19:
First step:
verify that everything works using CLI commands:

Start the node
`$ roslaunch zed_wrapper zedm.launch`
Open a terminal and subscribe to image
`$ rostopic hz /zed/zed_node/rgb/image_rect_color`
Open a terminal and subscribe to pose
`$ rostopic hz /zed/zed_node/pose`

Let me know the results

WondesenB, 2019-07-19:
Thank you Walter,  here  I put the results as you said.
[image]
[image]

Myzhar, 2019-07-19:
I can't see the resolution of the images, but it is not possible to get the pose at 60 Hz on the Jetson TX2 while performing other operations. It has not enough computational power.
I usually get the pose at  30 Hz (even 15 Hz) that is enough for a robotic application

WondesenB, 2019-07-19:
Thank you so much Walter, I am using it for drone indoor navigation , I set the camera resolution to HD720 at 60 fps. I test it for position hold and it is stable but with position subscription only. The problem comes when I start subscribing to image topic from camera, the system will become quite unstable (better with VGA resolution but not enough). would you recommend me better SBC (high computation power) ? Thanks a lot !

Myzhar, 2019-07-19:
Better of the TX2 with CUDA capabilities there is only the Xavier. 
If it is within your budget, it is surely a big step forward.

WondesenB, 2019-07-20:
Thank you a lot Walter for your help. Wish you all the best !

closed 2020-03-23 by Myzhar