stereolabs/zed-ros-wrapper #453: Zed unable to launch on boot using robot_upstart. Fails to detect Nvidia graphics card.

darshit4777, 2019-07-24:
I am trying to launch the zed camera on boot using robot_upstart. The service tries to launch a single launch file : zedm.launch. This fails as Zed is unable to detect a CUDA capable device and gives the error No NVIDIA graphics card detected.

The output from the terminal is as follows

Jul 24 15:01:50 loggerbox3 zed_camera-start[31592]: /usr/bin/du: cannot read directory '/tmp/systemd-private-860e9265709142ea9a7428aee816
fresh@loggerbox3:~/catkin_ws/src$ sudo systemctl status zed_camera.service
● zed_camera.service - "bringup zed_camera"
   Loaded: loaded (/lib/systemd/system/zed_camera.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2019-07-24 15:01:48 PDT; 2min 22s ago
 Main PID: 31592 (zed_camera-star)
    Tasks: 34 (limit: 4174)
   CGroup: /system.slice/zed_camera.service
           ├─31592 /bin/bash /usr/sbin/zed_camera-start
           ├─31692 /usr/bin/python /opt/ros/melodic/bin/roslaunch /tmp/zed_camera.launch
           ├─31703 /usr/bin/python /opt/ros/melodic/bin/rosmaster --core -p 11311 -w 3 __log:=/tmp/a3ae3f76-ae5e-11e9-89d5-00044be5579a/m
           ├─31714 /opt/ros/melodic/lib/rosout/rosout __name:=rosout __log:=/tmp/a3ae3f76-ae5e-11e9-89d5-00044be5579a/rosout-1.log
           ├─31717 /opt/ros/melodic/lib/robot_state_publisher/state_publisher robot_description:=zed_description __name:=zed_state_publis
           └─31718 /home/fresh/catkin_ws/devel/lib/zed_wrapper/zed_wrapper_node __name:=zed_node __log:=/tmp/a3ae3f76-ae5e-11e9-89d5-0004

Jul 24 15:04:05 loggerbox3 zed_camera-start[31592]: in Camera.cpp : void sl::Camera::close() at 204 : cuda error [38]: no CUDA-capable de
Jul 24 15:04:05 loggerbox3 zed_camera-start[31592]: No NVIDIA graphics card detected. Install an NVIDIA GPU, CUDA and restart your comput

I am trying to run this on a Jetson Nano. Also I am able to successfully launch zed and run its basic functionality using launch files. I am just not able to do so using robot_upstart.

The command line used to install the service is 

> rosrun robot_upstart install --job zed_camera --interface eth0 zed_wrapper/launch/zedm.launch

Myzhar, 2019-07-24:
Hi @darshit4777,
The problem with robot_upstart is that it starts the nodes when the X server is not yet loaded and so the GPU is not yet available.
I know that @rbonghi was working with it and had the same problem, but I do not know if he found a solution.

darshit4777, 2019-07-24:
Thanks @Myzhar  for that super quick response.

If that is the case, the systemd scripts could be modified to start my ZED service after the X server is up. I will try this. It'll be great to know if @rbonghi has anything to share from his experience.

darshit4777, 2019-07-24:
Actually @Myzhar on second thoughts, that is probably not the source of the problem.  I install the the ZED launch file as a systemd service after the entire system (including the X server) is up and immediately say, (without  rebooting)

```
systemctl start zed_camera.service
```
I get the same errors as above.

Myzhar, 2019-07-24:
Another thing you must take care of is that every user that needs to access the GPU must be added to the `video` group... if I remember well robot_upstart runs as `root`, so the `root` user must be added to the `video` group and I do not know if this is possible.

darshit4777, 2019-07-24:
I actually did try adding root to the group video. It is possible. Although it doesn't solve the issue.
The output for 
`cat /etc/group | grep -i video` 
is 
`video:x:44:fresh,gdm,root` 
Unfortunately that's exactly what was recommended on an Nvidia forum in 2016.
[https://devtalk.nvidia.com/default/topic/950027/jetson-tk1/zed-cuda-and-ros-as-an-upstart-background-process-upon-boot/](url)

rbonghi, 2019-07-25:
Hi @darshit4777 ,

Yes, I fixed, but the problem is in this line of `robot_upstart` 
https://github.com/clearpathrobotics/robot_upstart/blob/kinetic-devel/templates/job-start.em#L107

When roslaunch start make `setuidgid` like the owner of the file, but does not set all secondaries group where is located the user.
Second point the GPU is installed with privileged for user `video` and like the serial port `dialout` follow this rule https://answers.ros.org/question/207892/robot_upstart-could-not-open-port-devttyusb0/

I found two way to fix this issue:
1. Change the line with
```
roslaunch $LAUNCH_FILENAME @(roslaunch_wait?'--wait ')&
```
and roslaunch start with sudo
2. (used) change the primary group of your user and make primary in `video`

My experience with ubuntu is not high level, I don't know if is it possible to run with user and same primary and secondary level of group

rbonghi, 2019-07-25:
> I actually did try adding root to the group video. It is possible. Although it doesn't solve the issue.
> The output for
> `cat /etc/group | grep -i video`
> is
> `video:x:44:fresh,gdm,root`
> Unfortunately [...]

I'm afraid to make same level the user video.
Anyway is another dirty way to fix this issue

darshit4777, 2019-08-06:
I managed to fix this and all of the solutions work.

1. As @rbonghi said, removing `setuidgid` works because the process continues to run as root. But I didn't want that to happen.

2. Again like @rbonghi said, changing the primary group of the user to `video` also fixes the the issue because `setuidgid` does set the primary group but not supplementary group. But this was also not ideal for us because of other groups the user needed to be a part of.

3. I ended up using other re-implementations of `setuidgid` that do set supplementary groups. For example [daemontools-encore](https://untroubled.org/daemontools-encore/setuidgid.8.html). I compiled the package and copied the `setuidgid` executable as `setuidgid-encore` (to make sure other programs are not affected) and changed the `job-start.em` appropriately. 

Thanks!

closed 2019-08-06 by darshit4777