stereolabs/zed-ros-wrapper #454: How can we perform odometry under already created map?

icsl-Jeon, 2019-07-25:
Hello, thank you for great works. 

I was wondering I can start odometry from an already created map from a package such as rtabmap with boosted performance. 

According to your package, there is a parameter ```odometry_db``` but I could not find the details on it.   

As I see it, I think I can do what I want by specifying the .db file to  ```odometry_db``` and start another live odometry. Also, I think it might outperform the odometry from scratch. 

Could you show me the way? 

Best regards

Myzhar, 2019-07-25:
Hi @icsl-Jeon ,
you are saying it right, but you must record the map using the same `odometry_db` file.
The module we are speaking about is called "Spatial memory" and you can find the documentation on the Stereolabs website:
https://www.stereolabs.com/docs/positional-tracking/spatial-memory/

The ROS wrapper sets the DB using the `sl::TrackingParameters::area_file_path` parameter initialized with `odometry_db` value.

At the current state the ZED Wrapper can use only an existing database and cannot update it.

The feature will be added in a future release.

icsl-Jeon, 2019-07-25:
Thank you for your quick reply, Myzhar. 
1)
As I understand, your saying is zed_wrapper can only load an existing ```*.area``` but cannot create those file.  I think I have to make .area file from source code. 
2) 
If I try to use .area file by setting ```odometry_db``` parameters, are there any best practice for *"warm start"* of odometry? (e.g. start from distinct feature region in *.area file)

Best regards

Myzhar, 2019-07-25:
1) Yes. Follow the example I linked above
2) The tracking module searches for a know position in the DB automatically

icsl-Jeon, 2019-07-25:
Thank you so much! I am closing this thread.

Myzhar, 2019-07-25:
You are welcome.

I keep the issue open as a reminder to add the DB creation feature

closed 2022-04-19 by github-actions[bot]