stereolabs/zed-ros-wrapper #456: Ros Wrapper On Melodic?

JackFields, 2019-07-31:
Is there support for Ros Wrapper on Melodic? I tried following the tutorial but had no luck, I can get it running on kinetic fine. Is there anything I should do differently when installing it?

Myzhar, 2019-08-01:
Hi @JackFields 

The ZED ROS Wrapper works with ROS Melodic without any modification.
What kind of problem are you facing?

JackFields, 2019-08-01:
Hi,

When trying to install the wrapper, it returns an error since it is looking for the folders titled sl, sl_core, and sl_zed in the directory catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/tools/include. I did find these folders in /usr/local/zed/include and copy-pasted them over. However, in the sl_core/utils folder there is a file called types.hpp which has 4 more #include statements to:

#include <cuda.h>
#include <cuda_runtime.h>
#include <cuda_runtime_api.h>
#include <device_launch_parameters.h>

none of which I could find. What is it that I'm doing wrong?

Myzhar, 2019-08-01:
Hi @JackFields ,
this error seems not related to ROS, but to a bad configuration of the ZED SDK.

Can you please copy and paste the full output of the `catkin_make` command so I can analyze it deeper and give a more precise solution to your problem?

JackFields, 2019-08-01:
Hi,

I was able to fix the issue. the problem was that Cuda wasn't the right version. However, a new problem seems to have popped up. When I try to run display_zed.launch, the following error appears:

Resource not found: zed_wrapper
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/opt/ros/melodic/share

Myzhar, 2019-08-02:
If `catkin_make` was successfully completed then you have a problem with the path.
Have you sourced the configuration scripts?
Ros Melodic installation:
`$ source /opt/ros/melodic/setup.bash`
Catkin workspace:
`$ source ~/catkin_ws/devel/setup.bash` <- use the correct path for you

JackFields, 2019-08-02:
Thanks, the camera runs fine now.

closed 2019-08-02 by JackFields

sreshu, 2020-08-20:
[image]