stereolabs/zed-ros-wrapper #459: Jetson Nano D435 ROS wrapper | Error  ZED connection -> CAMERA NOT DETECTED

RatheeshRK, 2019-08-17:
Error while using zed-ros-wrapper:  ZED connection -> CAMERA NOT DETECTED
Hardware: NVIDIA Jetson Nano
Ubuntu 18.04, ROS melodic

The camera works with Jetson Nano with ''realsense-viewer'' (all features, with out any error)

Tried to follow the zed-ros-wrapper, but camera not detecting.

```
roslaunch zed_display_rviz display_zed.launch 
... logging to /home/rrk/.ros/log/f139bd78-c0de-11e9-922e-00044be415aa/roslaunch-JetsonNano-19578.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://JetsonNano:37677/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /zed/zed_description: <?xml version="1....
 * /zed/zed_node/auto_exposure: True
 * /zed/zed_node/camera_model: zed
 * /zed/zed_node/confidence: 100
 * /zed/zed_node/depth/confidence_root: confidence
 * /zed/zed_node/depth/depth_stabilization: 1
 * /zed/zed_node/depth/depth_topic_root: depth
 * /zed/zed_node/depth/disparity_topic: disparity/dispari...
 * /zed/zed_node/depth/min_depth: 0.3
 * /zed/zed_node/depth/openni_depth_mode: 0
 * /zed/zed_node/depth/point_cloud_topic_root: point_cloud
 * /zed/zed_node/depth/quality: 1
 * /zed/zed_node/depth/sensing_mode: 0
 * /zed/zed_node/exposure: 100
 * /zed/zed_node/gain: 100
 * /zed/zed_node/general/base_frame: base_link
 * /zed/zed_node/general/camera_flip: False
 * /zed/zed_node/general/camera_frame: zed_camera_center
 * /zed/zed_node/general/frame_rate: 30
 * /zed/zed_node/general/gpu_id: -1
 * /zed/zed_node/general/left_camera_frame: zed_left_camera_f...
 * /zed/zed_node/general/left_camera_optical_frame: zed_left_camera_o...
 * /zed/zed_node/general/resolution: 2
 * /zed/zed_node/general/right_camera_frame: zed_right_camera_...
 * /zed/zed_node/general/right_camera_optical_frame: zed_right_camera_...
 * /zed/zed_node/general/self_calib: True
 * /zed/zed_node/general/serial_number: 0
 * /zed/zed_node/general/svo_compression: 4
 * /zed/zed_node/general/verbose: True
 * /zed/zed_node/general/zed_id: -1
 * /zed/zed_node/mapping/fused_pointcloud_freq: 1.0
 * /zed/zed_node/mapping/mapping_enabled: False
 * /zed/zed_node/mapping/resolution: 1
 * /zed/zed_node/mat_resize_factor: 1.0
 * /zed/zed_node/max_depth: 20.0
 * /zed/zed_node/point_cloud_freq: 10.0
 * /zed/zed_node/stream: 
 * /zed/zed_node/svo_file: 
 * /zed/zed_node/tracking/fixed_cov_value: 1e-6
 * /zed/zed_node/tracking/fixed_covariance: False
 * /zed/zed_node/tracking/fixed_z_value: 1.0
 * /zed/zed_node/tracking/floor_alignment: False
 * /zed/zed_node/tracking/init_odom_with_first_valid_pose: True
 * /zed/zed_node/tracking/initial_base_pose: [0.0, 0.0, 0.0, 0...
 * /zed/zed_node/tracking/map_frame: map
 * /zed/zed_node/tracking/odometry_db: 
 * /zed/zed_node/tracking/odometry_frame: odom
[... 118 lines cut ...]
ZED (Init) >> Depth mode: PERFORMANCE
** Opening Camera. Attempt 0 ...
** Opening Camera. Attempt 1 ...
** Opening Camera. Attempt 2 ...
** Opening Camera. Attempt 3 ...
** Opening Camera. Attempt 4 ...
** Opening Camera. Attempt 5 ...
** Opening Camera. Attempt 6 ...
** Opening Camera. Attempt 7 ...
** Opening Camera. Attempt 8 ...
** Opening Camera. Attempt 9 ...
** Opening Camera. Attempt 10 ...
** Opening Camera. Attempt 11 ...
** Opening Camera. Attempt 12 ...
** Opening Camera. Attempt 13 ...
** Opening Camera. Attempt 14 ...
** Opening Camera. Attempt 15 ...
** Opening Camera. Attempt 16 ...
** Opening Camera. Attempt 17 ...
** Opening Camera. Attempt 18 ...
** Opening Camera. Attempt 19 ...
** Opening Camera. Attempt 20 ...
[ INFO] [1566039945.600693005]: ZED connection -> CAMERA NOT DETECTED

```

Myzhar, 2019-08-17:
Hi @RatheeshRK 
This is a ROS repository for the Stereolabs ZED and ZED mini cameras:
https://www.stereolabs.com/
You cannot use it with Intel products.

closed 2019-08-17 by Myzhar

zoombinis, 2020-12-09:
@Myzhar I do not understand your reason for closing this issue. I too am getting this same exact issue. The Zed camera only connects if an X11 server is running:
https://github.com/stereolabs/zed-ros-wrapper/issues/166#issuecomment-741962144

Myzhar, 2020-12-09:
@zoombinis this is not valid issue because @RatheeshRK was trying to use the ZED wrapper with an Intel device

zoombinis, 2020-12-09:
@Myzhar I'm sorry it's not apparent what that means, he's using the same hardware/OS as me:

> Hardware: NVIDIA Jetson Nano
Ubuntu 18.04, ROS melodic

Where is the "Intel" product and are you saying Zed can't be used on Jetson products (by NVidia?)

Also, possible you can comment on my linked issue? I can only seem to operate the Zed with X11 authenticated to my Docker Client running the Zed SDK.