stereolabs/zed-ros-wrapper #460: URDF Model of ZED Camera not displayed in rviz

mtbsteve, 2019-08-19:
I have the ZED ROS Wrapper installed on ROS melodic and Ubuntu 18.04.
Everything works fine as I can see, however I am unable to display the URDF model of the ZED camera in the rviz viewer.
When I launch rviz with:
`roslaunch zed_display_rviz display_zed.launch`
the camera model shows up in the righthand rviz window for a split second and disappears when the axes are displayed. The models are correctly installed under 
/home/.../ws/src/zed-ros-wrapper/zed_wrapper/urdf
Any idea?

Myzhar, 2019-08-19:
Hi @mtbsteve 
if you are using a ZED camera (not ZED Mini) this is strange since it works with no problems with us.
Have you modified any paramter in the YAML files or in the URDF description?

If you are instead using a ZED Mini camera you must call the other launch file:
`roslaunch zed_display_rviz display_zedm.launch`

mtbsteve, 2019-08-19:
thanks for getting back to me. 
I am using the ZED, not the mini camera. 
I have not modified any parameters in the YAML files nor in the URDF description. 
There are no error messages when I run rviz, and I couldn’t spot any errors in the log files. All examples as far as I tested them in the Stereolabs ROS tutorials seem to work. However I do experience some not consistently reproducible crashes of rviz from time to time when I add and remove ZED plugins.

closed 2020-02-18 by Myzhar