stereolabs/zed-ros-wrapper #461: offline generation of pointcloud

ben0110, 2019-08-21:
HI,

I'm collecting data with the zed camera through ROS. And in order to don't loose packets, my idea was to collect the images from the right and the left cameras and to generate offline the point clouds. is there some function that do that in the SDK that get as inputs the left and right images and generate a pointcloud message?

Myzhar, 2019-08-21:
Hi @ben0110,
You can record an SVO using the ZED Explorer tool (or directly using the ROS wrapper), then you can launch the ROS wrapper choosing the SVO as input source.
You can find more info here:
https://www.stereolabs.com/docs/video/#video-recording
https://www.stereolabs.com/docs/ros/#launching-with-recorded-svo-video

ben0110, 2019-10-15:
Hi @Myzhar,

In my case, Using SVO files is not an option, as i'm simultaneously collecting data from other sensors.

closed 2019-10-15 by ben0110