stereolabs/zed-ros-wrapper #462: Drift in odometry using zed mini

gsh89, 2019-08-22:
I am using a zed mini camera. When I view the /odom topic, the x and y position values change a lot even on a stationary robot. The y value fluctuates about zero and the x keeps on increasing. Any ideas why this would be ?

gsh89, 2019-08-26:
Hi @Myzhar. Do you need more information for providing help on this ?

Myzhar, 2019-08-26:
Hi @gsh89,
yes, I do. It seems like if your IMU is not well calibrated and there is a little bias.
Can you provide me with a log of the IMU RAW topic?

Myzhar, 2019-08-26:
A text file for the topic is good. You can simply copy and paste the output of the command (the ZED node should be running):
`$ rostopic echo /zed/zed_node/imu/data_raw`
You can find the version of the SDK you are using as console output when the ZED node starts.

laranjma, 2019-10-11:
Hi @Myzhar, I have the same problem, the IMU drifts and I cannot use it to reference my measurements in the Earth frame. Here below a copy and paste of imu/data_raw. I am using the latest SDK version (2.8.3)
Thanks
---
header: 
  seq: 856
  stamp: 
    secs: 1570806998
    nsecs: 210817457
  frame_id: "zed_imu_link"
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: -0.00479369009399
  y: 0.001597896698
  z: 0.00532632232666
angular_velocity_covariance: [4.527458679038042e-06, 0.0, 0.0, 0.0, 4.91844405810675e-06, 0.0, 0.0, -0.0, 5.788329709098867e-06]
linear_acceleration: 
  x: 1.27530634403
  y: 0.0743129625916
  z: 9.75417613983
linear_acceleration_covariance: [0.0015851784264668822, 0.0, 0.0, 0.0, 0.0013550298754125834, 0.0, 0.0, -0.0, 0.017072943970561028]
---
header: 
  seq: 857
  stamp: 
    secs: 1570806998
    nsecs: 210817457
  frame_id: "zed_imu_link"
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: -0.00106526446533
  y: -0.000532632232666
  z: 0.00692421902466
angular_velocity_covariance: [4.527458679038042e-06, 0.0, 0.0, 0.0, 4.91844405810675e-06, 0.0, 0.0, -0.0, 5.788329709098867e-06]
linear_acceleration: 
  x: 1.31366145611
  y: 0.0719157680869
  z: 9.76616191864
linear_acceleration_covariance: [0.0015851784264668822, 0.0, 0.0, 0.0, 0.0013550298754125834, 0.0, 0.0, -0.0, 0.017072943970561028]
---
header: 
  seq: 858
  stamp: 
    secs: 1570806998
    nsecs: 210817457
  frame_id: "zed_imu_link"
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: -0.00585895455933
  y: -0.00852211572266
  z: 0.00479369009399
angular_velocity_covariance: [4.527458679038042e-06, 0.0, 0.0, 0.0, 4.91844405810675e-06, 0.0, 0.0, -0.0, 5.788329709098867e-06]
linear_acceleration: 
  x: 1.26571762562
  y: 0.122256807983
  z: 9.75657272339
linear_acceleration_covariance: [0.0015851784264668822, 0.0, 0.0, 0.0, 0.0013550298754125834, 0.0, 0.0, -0.0, 0.017072943970561028]
---
header: 
  seq: 859
  stamp: 
    secs: 1570806998
    nsecs: 210817457
  frame_id: "zed_imu_link"
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: -0.00639158679199
  y: -0.0213052893066
  z: 0.00213052893066
angular_velocity_covariance: [4.527458679038042e-06, 0.0, 0.0, 0.0, 4.91844405810675e-06, 0.0, 0.0, -0.0, 5.788329709098867e-06]
linear_acceleration: 
  x: 1.31605863571
  y: 0.115065231919
  z: 9.72540950775
linear_acceleration_covariance: [0.0015851784264668822, 0.0, 0.0, 0.0, 0.0013550298754125834, 0.0, 0.0, -0.0, 0.017072943970561028]
---
^Cheader: 
  seq: 860
  stamp: 
    secs: 1570806998
    nsecs: 210817457
  frame_id: "zed_imu_link"
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: -0.00745685125732
  y: -0.0197073946892
  z: -0.000532632232666
angular_velocity_covariance: [4.527458679038042e-06, 0.0, 0.0, 0.0, 4.91844405810675e-06, 0.0, 0.0, -0.0, 5.788329709098867e-06]
linear_acceleration: 
  x: 1.2585259676
  y: 0.08390173316
  z: 9.75657272339
linear_acceleration_covariance: [0.0015851784264668822, 0.0, 0.0, 0.0, 0.0013550298754125834, 0.0, 0.0, -0.0, 0.017072943970561028]
---

Myzhar, 2020-03-23:
Positional tracking has been improved with the latest version of the SDK.
I close the issue

closed 2020-03-23 by Myzhar