stereolabs/zed-ros-wrapper #464: Corruped SVO files

changho288, 2019-08-29:
I have some svo files recored with an incomplete process such as computer powered off and the recording program forced shutdown. I think it caused some header issues. The files have several GB sizes, so I believe those contains video data. Please give me some advise or information to repair those. Thank you.

Myzhar, 2019-08-29:
Hi @changho288 
You can try to recover it using the `ZED SVOEditor` with the `--repair` option

changho288, 2019-08-30:
Thank you so much for your answering.
Could you tell me how can I get ZED SVOEditor or some example codes? I am currently using ZED SDK v2.4 but I do not have it.

Myzhar, 2019-08-30:
You can find the executable in the Tools folder: `/usr/local/zed/tools`.
If it is not present you can install a newer version of the SDK.
The v2.4 is really old, I suggest you to install the latest v2.8.3

changho288, 2019-09-02:
[image]

I recently setup the newest version of ZED SDK, but only three tools, Depth Viewer, Explorer, and ZEDfu, have been installed in the Tools folder like the attached picture.

obraun-sl, 2019-09-02:
The tools are located in C:/Program Files (x86) /ZED SDK/tools/
What you are showing in the picture is the start Menu folder with only contains shortcuts for some of the tools.

Myzhar, 2019-09-02:
@changho288 since this is not a ROS related issue, you are out of topic in this repository.
If you have further questions and you require other kind of support please write an email to support@stereolabs.com.
Thank you
Walter

closed 2019-09-02 by Myzhar

changho288, 2019-09-02:
Thank you obraun-sl. I found it. 

I just linked in this section with searching of SVO recording issue. I did not know this section was the ROS one. I am sorry about that. Thank you for your concern Myzhar.