stereolabs/zed-ros-wrapper #466: Zed Mini CPU Usage

sinankusdogan, 2019-09-05:
We are currently using Zed Mini on Ubuntu 16.04 with i7 CPU, 16 GB RAM and GTX 1050 Ti GPU. We are first trying to subscribe only to imu topic. The zed_wrapper node uses about 32% CPU. Is it normal or shall we check some parameters to reduce this usage? Resolution is VGA, FPS is 15. We've tried Raw, H264 & H265 compressions and that made no difference.

Myzhar, 2019-09-05:
Hi @sinankusdogan,
Are you only subscribing the IMU topic or also other topics?

To publish the IMU topic the node creates a parallel thread to the main thread and it requires a few calculation power depending on the requested IMU frequency.

32% of the CPU can have a lot of meaning, how do you measure it?

sinankusdogan, 2019-09-05:
Yes, i am only subscribing to imu/data topic. 

I am running "top" process on Ubuntu to see CPU usage. It looks like very high compared to other nodes as move_base or localization or mapping nodes. Also system slows down when i run zed_wrapper. I thought it would mostly use GPU instead of CPU.

Myzhar, 2019-09-06:
The GPU is used by the ZED SDK to perform depth calculation and other heavy data processing.
The ZED node take the data and prepare them to be distributed in the ROS world.

I have three other questions for you:
1) What is the version of the ZED SDK that you are using?
2) Are you using the latest ZED node available in the master branch?
3) What is the frequency of the IMU topic?

sinankusdogan, 2019-09-06:
1. ZED SDK is the latest version (2.8). 
2. We have pulled lthe atest codes for zed_wrapper also. 
3. Tried different frequencies as 100, 50, 20. 

It was about 45% at the beginning and it has been decreased to 32% by changing resolution, frame_rate and disabling of publishing tf. But i think it should be as 10% or 15% at most. I know it's very relative but according to my experiences and compared to other nodes as navigation, controllers etc. 32% seems like very high.

Myzhar, 2019-09-06:
Can you try to set IMU frequency to 800 Hz that is the maximum allowed value and check if the value increases or decreases?

However you cannot compare the ZED node power requirements to other nodes. Each node acts in a totally different way

sinankusdogan, 2019-09-06:
It has been increased to 44%.

Myzhar, 2019-09-06:
Can you check the status of the single cores of your CPU? 
I expect that one core is running at 100%

sinankusdogan, 2019-09-06:
I may have wasted your time. Thank you for your help. 

When looking to CPU time over all possible CPU power (pressing shift + i), it becomes 5.5%. For additional information, cores are loaded equally. No one is 100%. It seems well now. Slow performance issue may also been related to network. I will check it. I am closing the issue. Have a nice day.

closed 2019-09-06 by sinankusdogan