stereolabs/zed-ros-wrapper #467: Playing back svo file

AndinetH, 2019-09-09:
I have recorded an svo video using zed camera. I am wondering if there is an option to play the video at a lower rate than it was recorded. The command I am using is as follows:
roslaunch zed_wrapper zed.launch svo_fie:="/path/to/file/HD2K_xxx.svo".

Thanks

Myzhar, 2019-09-09:
Hi @AndinetH 
the ZED node does not allow to change the speed of the SVO playback since it is designed to replicate the real FPS of the camera that has beel used while recording it.

You can however modify the node to get what you are searching for:
- verify that "real time mode" is not active: https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp#L141
- change the rate of the main thread of the node (`device_poll_thread_func`) modifying the value of the parameter `mCamFrameRate`: https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp#L2077

AndinetH, 2019-09-10:
Thank you for the detailed explanation. I think the mCamFrameRate variable could be any number, not just 15/30/60/100 fps available from the zed explorer tool. One more question: is the sample source file that you provide at https://github.com/stereolabs/zed-examples/tree/master/svo%20recording/recording a command line equivalent to recording using the zed explorer? thanks.

Myzhar, 2019-09-10:
15/30/60/100 fps are the values allowed by the camera sensors, that's why they are fixed.
Playing the SVO in "slow motion" is not a common request and modifying the node to get it is the better way to not confuse other users.

The recording example is exactly what you said.

If you want you can also use the ZED node to directly record an SVO simply using the "start_svo_recording" service described here: https://www.stereolabs.com/docs/ros/zed_node/#services

AndinetH, 2019-09-10:
Thank you so much.

closed 2019-09-10 by AndinetH