stereolabs/zed-ros-wrapper #468: Handeye calibration of Zed Mini

howard-wu-cam, 2019-09-10:
I am fairly new on Zed Mini development, apologies if it is a repeated question. **Has anyone done eye-on-base calibration of Zed Mini?** I have looked around, but can't find any related information. I am not looking for a detailed solution, but it would be extremely helpful if someone can point me to the right direction. Not sure whether this is the right place to post the issue either, but since I am working with ROS, so I hope it is not totally out of the scope. Thanks!

Myzhar, 2019-09-10:
Hi @howard-wu-cam 
Do you have any link to better understand what you mean with `eye-on-base calibration`?

howard-wu-cam, 2019-09-10:
Thank you @Myzhar , here is a [link](https://github.com/IFL-CAMP/easy_handeye) to explain what is handeye calibration, including both eye-on-base and eye-in-hand calibration.

Myzhar, 2020-03-23:
Closed fo inactivity

closed 2020-03-23 by Myzhar