stereolabs/zed-ros-wrapper #469: Slow publishing rate of zed images

tianshiz, 2019-09-10:
On my local computer I can get 30 Hz while checking with rostopic hz ../rgb/image_rect_color

However, when looking at the same topic from another computer in the LAN, connected via 5Ghz WiFi, the hz drops to 1-2 Hz.

Is this normal? I'm also using the point cloud. Is it just my network is too congested? Does ROS automatically throttle it? I'm streaming 720p with SVO compression.

Is there a workaround to get the topics to show with decent fps?

Myzhar, 2019-09-10:
Hi @tianshiz 
if you repeat the test using a cable LAN you will notice that everything works like expected.

The bandwidth of wifi is not enough to handle such an amount of traffic, you should reduce it.

You can try to reduce the BW of the pointcloud topic using the 'throttle' node in the 'topic_tools' stack: http://wiki.ros.org/topic_tools/throttle (yes, ROS does not handle throttle automatically).

Another way to reduce bandwidth without losing precision in the ZED SDK algorithm is using the "mat_resize_factor" dynamic parameter in the ZED node: https://www.stereolabs.com/docs/ros/zed_node/#zed-parameters

Let me know if this helps

tianshiz, 2019-09-10:
Hi @Myzhar,

 Thank you for the help! What's the difference between setting depth quality to "performance" vs using topic_tools?

 Apologies but I'm a little unsure about what mat_resize_factor does, do you mind explaining that for me?

Myzhar, 2019-09-10:
Using "topic_tools" you reduce the number of transmitted frames.
Changing the quality (reducing the frame size) you keep constant the frame rate.

"mat_resize_factor" is a kind of "quantization" of the data. You reduce the bandwidth reducing the size of the single frame.

You could also use compressed frames for image topics, but it requires a lot of computational power

closed 2020-02-18 by Myzhar