stereolabs/zed-ros-wrapper #472: catkin_make error for zed wrapper

ghost-60, 2019-09-21:
Zed sdk: 2.4.1
Ubuntu: 18.04
nvcc --version:
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2017 NVIDIA Corporation
Built on Fri_Nov__3_21:07:56_CDT_2017
Cuda compilation tools, release 9.1, V9.1.85

I am getting following error when I run catkin_make in root folder:

Base path: /home/ghost/ghost2
Source space: /home/ghost/ghost2/src
Build space: /home/ghost/ghost2/build
Devel space: /home/ghost/ghost2/devel
Install space: /home/ghost/ghost2/install
####
#### Running command: "make cmake_check_build_system" in "/home/ghost/ghost2/build"
####
####
#### Running command: "make -j4 -l4" in "/home/ghost/ghost2/build"
####
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target _beginner_tutorials_generate_messages_check_deps_AddTwoInts
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target _beginner_tutorials_generate_messages_check_deps_Num
[  0%] Built target std_msgs_generate_messages_eus
[  2%] Built target tf_listener
[  5%] Built target zed_depth_sub
[  7%] Built target tf_broadcaster
[ 10%] Built target zed_tracking_sub
[ 13%] Built target zed_video_sub
[ 13%] Built target _zed_wrapper_generate_messages_check_deps_stop_remote_stream
[ 13%] Built target _zed_wrapper_generate_messages_check_deps_start_svo_recording
[ 13%] Built target _zed_wrapper_generate_messages_check_deps_stop_svo_recording
[ 13%] Built target _zed_wrapper_generate_messages_check_deps_toggle_led
[ 13%] Built target _zed_wrapper_generate_messages_check_deps_reset_odometry
[ 13%] Built target _zed_wrapper_generate_messages_check_deps_reset_tracking
[ 14%] Built target _zed_wrapper_generate_messages_check_deps_set_led_status
[ 14%] Built target zed_wrapper_gencfg
[ 14%] Built target _zed_wrapper_generate_messages_check_deps_set_pose
[ 17%] Built target beginner_tutorials_generate_messages_cpp
[ 19%] Built target beginner_tutorials_generate_messages_nodejs
[ 22%] Built target beginner_tutorials_generate_messages_lisp
[ 22%] Built target _zed_wrapper_generate_messages_check_deps_start_remote_stream
[ 27%] Built target beginner_tutorials_generate_messages_py
[ 31%] Built target beginner_tutorials_generate_messages_eus
[ 43%] Built target zed_wrapper_generate_messages_lisp
[ 55%] Built target zed_wrapper_generate_messages_nodejs
[ 68%] Built target zed_wrapper_generate_messages_py
[ 80%] Built target zed_wrapper_generate_messages_cpp
[ 93%] Built target zed_wrapper_generate_messages_eus
[ 93%] Built target beginner_tutorials_generate_messages
[ 93%] Built target zed_wrapper_generate_messages
[ 94%] Building CXX object zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/src/nodelet/src/zed_wrapper_nodelet.cpp.o
/home/ghost/ghost2/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp: In member function ‘void zed_wrapper::ZEDWrapperNodelet::start_tracking()’:
/home/ghost/ghost2/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:1181:21: error: ‘struct sl::TrackingParameters’ has no member named ‘enable_imu_fusion’
         trackParams.enable_imu_fusion = mImuFusion;
                     ^~~~~~~~~~~~~~~~~
/home/ghost/ghost2/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp: In member function ‘void zed_wrapper::ZEDWrapperNodelet::publishOdom(tf2::Transform, sl::Pose&, ros::Time)’:
/home/ghost/ghost2/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:1224:70: error: ‘struct sl::Pose’ has no member named ‘pose_covariance’; did you mean ‘pose_confidence’?
                odom.pose.covariance[i] = static_cast<double>(slPose.pose_covariance[i]);
                                                                     ^~~~~~~~~~~~~~~
                                                                      pose_confidence
/home/ghost/ghost2/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp: In member function ‘void zed_wrapper::ZEDWrapperNodelet::publishPose(ros::Time)’:
/home/ghost/ghost2/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:1298:87: error: ‘struct sl::Pose’ has no member named ‘pose_covariance’; did you mean ‘pose_confidence’?
       poseCov.pose.covariance[i] = static_cast<double>(mLastZedPose.pose_covariance[i]);
                                                                     ^~~~~~~~~~~~~~~
                                                                                       pose_confidence
[... 83 lines cut ...]
         ^~~~~~~~~~
/home/ghost/ghost2/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp: At global scope:
/home/ghost/ghost2/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:2926:10: error: ‘ZEDWrapperNodelet’ has not been declared
     bool ZEDWrapperNodelet::on_start_remote_stream(zed_wrapper::start_remote_stream::Request& req,
          ^~~~~~~~~~~~~~~~~
/home/ghost/ghost2/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:2995:10: error: ‘ZEDWrapperNodelet’ has not been declared
     bool ZEDWrapperNodelet::on_stop_remote_stream(zed_wrapper::stop_remote_stream::Request& req,
          ^~~~~~~~~~~~~~~~~
/home/ghost/ghost2/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:3012:10: error: ‘ZEDWrapperNodelet’ has not been declared
     bool ZEDWrapperNodelet::on_set_led_status(zed_wrapper::set_led_status::Request& req,
          ^~~~~~~~~~~~~~~~~
/home/ghost/ghost2/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:3030:10: error: ‘ZEDWrapperNodelet’ has not been declared
     bool ZEDWrapperNodelet::on_toggle_led(zed_wrapper::toggle_led::Request& req,
          ^~~~~~~~~~~~~~~~~
/home/ghost/ghost2/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:3049:1: error: expected declaration before ‘}’ token
 } // namespace
 ^
zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/build.make:86: recipe for target 'zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/src/nodelet/src/zed_wrapper_nodelet.cpp.o' failed
make[2]: *** [zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/src/nodelet/src/zed_wrapper_nodelet.cpp.o] Error 1
CMakeFiles/Makefile2:4570: recipe for target 'zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/all' failed
make[1]: *** [zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
Please help resolve the issue

Myzhar, 2019-09-21:
Hi @ghost-60 
The ZED SDK v2.4.1 that you are using is very old and the compatibility with the latest version of the ROS wrapper is not guaranteed.
Please update to latest SDK v2.8.3, you will also get a lot of improvements and new features.

ghost-60, 2019-09-21:
My cuda version is 9.1 which zed sdk v2.8.3 does not support. Is there any other way I can resolve the issue?

Myzhar, 2019-09-21:
If you change CUDA version you must delete the cache before trying to re-compile.
Try removing the folders `devel` and `build` in your catkin workspace and then run `catkin_make` again

ghost-60, 2019-09-21:
Thanks it is working now~

closed 2019-09-21 by Myzhar