stereolabs/zed-ros-wrapper #473: Dynamic adjust point cloud publish frequency

jsseng, 2019-09-24:
Is there a parameter to allow run-time reconfiguration of the point cloud publish rate?  I see the /zed_nodelet_manager/parameter_updates topic, but it says it is publishing the point cloud at 15.0 Hz, when I am actually publishing at 8.0Hz (set in the .yaml config file).  I would like to turn on/off point cloud publishing at run-time to reduce CPU time.  Thanks.

Myzhar, 2019-09-24:
Hi @jsseng ,
pointcloud publishing is automatically turned off if no nodes subscribe to the pointcloud topic.

jsseng, 2019-09-24:
What if I wanted to adjust the point cloud publish rate though?  It would be nice to stay subscribed and set the publish rate to low, high, or even 0 Hz.

Myzhar, 2019-09-24:
You cannot. The pointcloud is always published at the camera frequency since it is calculated internally to the SDK processing, not by the ZED node.
CPU is only used to create the ROS topic from the ZED SDK data

jsseng, 2019-09-24:
It seems like the pointcloud computation is done in the SDK, but the publishing rate is controlled by the wrapper.  Could that publishing rate be made into a dynamic parameter?  

I was looking at pointcloud_thread_func() and the amount of time the thread sleeps is based on the publishing rate.

Myzhar, 2019-09-24:
@jsseng you are right, I forgot we've added the `point_cloud_freq` parameter that controls the pointcloud publishing frequency.
It is a dynamic parameter indeed: https://www.stereolabs.com/docs/ros/zed_node/#zed-parameters

jsseng, 2019-09-24:
Okay, it looks like I can adjust it there.  Thanks for the help!

jsseng, 2019-09-24:
Sorry to reopen.  I set the publish frequency in the .yaml config to 8Hz and when I check the frequency of the point cloud it is around 8Hz.

I called:  rosparam set /zed_nodelet_manager/point_cloud_freq 1.0  to set the frequency to 1.0 Hz.

When I call:  rosparam get /zed_nodelet_manager/point_cloud_freq, I do see the frequency has changed to 1.0 Hz.  But the cloud is still published at 8.0Hz.

It is as if the dynamic reconfigure callback is not using the new value.

Myzhar, 2019-09-24:
I check it and let you know

Myzhar, 2020-03-23:
Fixed with latest version of the wrapper

closed 2020-03-23 by Myzhar