stereolabs/zed-ros-wrapper #474: Error in displaying point cloud

mtbsteve, 2019-09-25:
I am trying to stream the node: /zed/zed_node/point_cloud/cloud_registered via rtsp to a client pc by using the ros_rtsp service (https://github.com/CircusMonkey/ros_rtsp)
That works well for the zed_nodes except point_cloud.
The error I am getting from zed_wrapper is:
`[ERROR] [1569393010.608909391]: Client [/standalone_nodelet] wants topic /zed/zed_node/point_cloud/cloud_registered to have datatype/md5sum [sensor_msgs/Image/060021388200f6f0f447d0fcd9c64743], but our version has [sensor_msgs/PointCloud2/1158d486dd51d683ce2f1be655c3c181]. Dropping connection. `

Point cloud is working in rviz, so I assume that the zed ros wrapper works so far. Any hints?
Thanks!

Myzhar, 2019-09-25:
Hi @mtbsteve 
I think that the error message is self- explaining: you cannot stream a pointcloud using RTSP, only images, indeed it requires a `sensor_msgs/Image` message type, while the pointcloud type is `sensor_msgs/PointCloud2`

mtbsteve, 2019-09-25:
Thanks and sorry for my ignorance :-)

closed 2019-09-25 by mtbsteve