stereolabs/zed-ros-wrapper #475: Question about ZED TF

tianshiz, 2019-09-27:
In reference to https://github.com/stereolabs/zed-ros-wrapper/issues/170

I'm slightly confused about the frames that I'm seeing. The camera_frame and and left_camera_frame are all in the X-out frame convention. Whereas the left_camera_optical_frame is in the Z-out per REP 103.

This is all fine and all, but ../point_cloud/cloud_registered is actually using the left_camera_frame instead of left_camera_optical_frame. 

Am I just missing a concept here? Shouldn't the point cloud also be in the Z-out frame? The messages I'm getting are all in the X-out frame.

Myzhar, 2019-09-29:
We could have chosen to publish it in the optical frame (Z-forward) but this requires than to transform it in camera frame (X-forward) both for visualization purposes (Rviz) and for obstacle detection and navigation.
If you like it to be in optical frame you can modify the source code such to publish it with Z-forward

tianshiz, 2019-09-30:
thanks for the clarification

closed 2019-09-30 by tianshiz