stereolabs/zed-ros-wrapper #476: Calculating X,Y,Z position of object from ROS topics

qboticslabs, 2019-09-30:
Hello

How can we extract the real-world coordinate (X, Y, Z) of an object w.r.t camera from the corresponding pixel coordinates of the RGB frame? 

We are using zed-ros-wrapper and doing object detection on RGB image, we want to find the real-world coordinates corresponding to the pixel coordinates.

Regards

Myzhar, 2019-09-30:
Hi @qboticslabs 
to extract XYZ real word coordinate you must keep in mind that the ZED SDK publishes the RGB image, the depth map/image and the point cloud as "registered" data, this means that at the pixel (u,v) of the RGB image corresponds exactly to the same pixel (u,v) of the depth image/map and to the same point (u,v) of the matrix containing the XYZ data of the point cloud.

This tutorial can be a starting point for your processing node:
https://github.com/stereolabs/zed-ros-wrapper/tree/master/tutorials/zed_depth_sub_tutorial

qboticslabs, 2019-09-30:
Thanks @Myzhar . I will check and get back.

closed 2019-09-30 by qboticslabs