stereolabs/zed-ros-wrapper #478: Doe the camera_flip param account for the frame_id of the pointcloud?

tianshiz, 2019-09-30:
I noticed that if I set camera_flip to true on a an upside down camera, the point cloud still follows the same axis convention(+X forward, +Z up). I'm assuming that's because zed is automatically correcting for this? It's convenient, but normally the frame_id is using the zed_left_camera, once flipped, it still is using that frame. 

So, my question is, what camera frame physically are we actually using in this case? Is it the right camera(when right-side up) that was flipped into the left side? In my urdf, the zed_center frame is shown to have rotated by 180, see below:
[image]

Myzhar, 2019-10-01:
Hi @tianshiz 
like you suppose the flipped camera is still using the same camera-sensor as reference, so the two pointclouds have and offset of 12 cm on X.
You could avoid this changing the "main" camera-sensor setting `enable_right_side_measure` to true in the structure `InitParameters`:
https://www.stereolabs.com/docs/api/structsl_1_1InitParameters.html#a3d125168204424b0dae2c317ca8804af

This option is not available in the ROS wrapper since it requires additional computational power to perform the "depth migration", but you can enable it to test if it does the work for you.

tianshiz, 2019-10-01:
so, correct me if I'm wrong, it seems like I could just flip the camera in reality, but in the URDF model pretend it's just upright. The only thing I would need to do is set the camera_flip value and this camera would operate like any other upright camera

Myzhar, 2019-10-01:
That's it 👍

closed 2019-10-01 by tianshiz