stereolabs/zed-ros-wrapper #480: FPS dropping when recording ZED data with rosbag

TBondat, 2019-10-03:
I'm working with ZED camera and ros kinetic on Ubuntu 16.04.
It appears that when i do rosbag record to store the data, the fps drop below 1fps(!) if i keep all topics.
I have an SSD drive with a Dell precision 7720 laptop, which is not too bad. I don't get problem when recording lidars + basic cameras for instance.

How can i improve the fps of ZED topics when recording a bag file? i already extended the --buffsize to 0 (infinite) but i gain like 1fps only. 
I tried degrading quality by switching to VGA@30 but still, if i keep all topics, it doesn't make the job.
Any hint for this problem? thanks in advance!

Myzhar, 2019-10-03:
Hi @TBondat 
I'm sorry but I don't really know how you could improve the recording.
Try to post the question on ROS answer since it seems more related to ROS than to the ZED wrapper itself: https://answers.ros.org/

Here at Stereolabs we suggest to record an SVO instead of a rosbag and to open the ZED wrapper with the SVO as input.

TBondat, 2019-10-03:
ok i see. Then i have 2 related questions:
- From the ROS ZED wrapper package, how can i launch a SVO recording? so programmatically, not via the ZED explorer?
- I didn't know the .svo format. Is it possible to synchronize a .svo file with a bag file that has been created at the same time? like by having shared the ROS timestamps.

Myzhar, 2019-10-03:
You can start a recording using the `start_svo_recording` service.
You can read more in the documentation: https://www.stereolabs.com/docs/ros/zed_node/#services

Currently it's not possible to sync an SVO with rosbag data. We are trying to find a solution to the "timestamp problem", but we have not yet succeded:
https://github.com/stereolabs/zed-ros-wrapper/issues/411

TBondat, 2019-10-04:
I agree this feature of synchronization svo-bag would be very useful.

To reduce the amount of data recorded, i guess it is possible to rebuild later the depth pointcloud, right? For that, which topics are used to build it exactly? I need the distance map and the rgb map.
Which topic name are they? There is a wide number of topics available.
Can it be /zed/zed_node/rgb/image_rect_color and /zed/zed_node/depth/depth_registered?

Myzhar, 2019-10-04:
They are 👍

TBondat, 2019-10-04:
Perfect! i have a last question. In .svo file used for recording, does it embed all topics or is it a selection of signals only, by default?
It is to know if with this recording format, i will have access to all the same topics available in ROS from ZED cam, later for postprocessing.

Myzhar, 2019-10-07:
The only missing information (currently) is the RAW inertial data

TBondat, 2019-10-10:
ok, thanks!

closed 2019-10-10 by TBondat

bramth, 2020-11-25:
I know this is an old issue but I just stumpled upon the same problem. What helped for me was to NOT use/save the depth/compressedDepth topic as it uses a lot of processing power (see http://wiki.ros.org/rtabmap_ros/Tutorials/RemoteMapping#Robot for more info.)