stereolabs/zed-ros-wrapper #484: Point_cloud_data and image_data have very slow sending rate with ROSBridge

eisenWu, 2019-10-15:
Hi, 

currently i am working with visualization of ZED_data, the camera works on JETSON TX2 and i use ROSBridge to send the cameradata to another operating system(windows 10), the problem which i have is:
i recived the point cloud data and image data from camera via ROSBridge, and the sending rata is like once per 1,5sec, which more confused the data is not the currently data it has a huge delay, i moved the camera manually and like 5 min later i recived the data.

Best regards 
Wei

Myzhar, 2019-10-15:
Hi @eisenWu 
are you using WiFi or a cabled network?
The pointcloud is very huge and sending it requires enough bandwidth according to data frequency and data resolution.

The 5 minute of latency is a clear signal of congested network.

You can try difference solutions:
1) reduce resolution
2) reduce FPS
3) use topic_throttle to send a minor amount of data
4) reduce the data dimensions playing with the `mat_resize_factor` dynamic parameter

eisenWu, 2019-10-15:
> 
> 
> Hi @eisenWu
> are you using WiFi or a cabled network?
> The pointcloud is very huge and sending it requires enough bandwidth according to data frequency and data resolution.
> 
> The 5 minute of latency is a clear signal of congested network.
> 
> You can try different solutions:
> 
>     1. reduce resolution
> 
>     2. reduce FPS
> 
>     3. use topic_throttle to send a minor amount of data
> 
>     4. reduce the data dimensions playing with the `mat_resize_factor` dynamic parameter

Hi @Myzhar 

thank you for the quick answer :)
i use the WebSocket namely Rosbridge to connet Unity and Ros, i will try your advises.
one more question, i am new to programming, as i read the [Doku](https://www.stereolabs.com/docs/video/using-video/#camera-configuration), i can adjust the camera configure with creating a camera object and with code in it, or i can just change the parameters in some current code.

thanks again!
Wei

Myzhar, 2019-10-15:
The documentation is relative to the SDK APIs. The ZED ROS Wrapper is an interface between the SDK APIs and the ROS world.

Unless you must add any kind of particular feature you don't need to change the ZED Wrapper code.

All the parameters are available in the yaml files in the `param` folder and you can change them to create your own configuration as explained here:
https://www.stereolabs.com/docs/ros/zed_node/#zed-parameters

eisenWu, 2019-10-16:
Hi @Myzhar,

i've tried reduce the resolution and the fps, it helps, thanks a lot.
The ROSBridge transfer the sensordata via JSON, it is still a huge amount of data, do u have any idea how to improve the performance with the transfer of data in real-time?
Wei

Myzhar, 2019-10-16:
Try to use the topic_throttle node to send not all the generated topics

eisenWu, 2019-10-17:
Hi @Myzhar ,

i will try with topic_throttle, but i am just newbee to ROS and so on, just need some time:)
Another question, since i changed the resolution of ZED to "VGA", i was wondering is it possible to use a customer resolution (like 320*240)?
Wei

Myzhar, 2019-10-17:
You cannot use a custom resolution, but you can reduce the size of the output data modifying the `mat_resize_factor` parameter.
It is a dynamic parameter so you can change it while the node is running to test it runtime

Myzhar, 2019-10-24:
Sure, you can set it in the file `common.yaml`:
https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/params/common.yaml

eisenWu, 2019-10-24:
Hi @Myzhar thanks a lot for your help:)

closed 2019-10-24 by eisenWu

milan-r-shah, 2019-11-14:
> Hi @eisenWu
> are you using WiFi or a cabled network?
> The pointcloud is very huge and sending it requires enough bandwidth according to data frequency and data resolution.
>
> The 5 minute of latency is [...]

Hi @Myzhar  Could you please let me know what is the difference between changing the resolution and changing the 'mat_size_factor' parameter? Because when I reduced the latter parameter to 0.1, the quality of PointCloud was considerably low.
Thanks in advance!

Myzhar, 2019-11-14:
The resolution impacts on the elaboration results because you change effectively the resolution of the image acquired by the camera sensors.
When you change the `mat_size_factor` you only modify what you see from the user side, but the SDK internally uses the resolution of the camera.

For example, if you set the camera at 1280x720 and you set the `mat_resize_factor` to 0.1 you will see images and point cloud as acquired by a 128x72 sensor, that's why the quality is low.

milan-r-shah, 2019-11-14:
> The resolution impacts on the elaboration results because you change effectively the resolution of the image acquired by the camera sensors.
> When you change the `mat_size_factor` you only modify [...]

In that case, just to confirm, if the 'mat_size_factor' only affects how a user sees an output, not the resolution used by the SDK then changing it would not affect the CPU usage right? Please correct me if I'm wrong.
Also, apart from capturing images, does ZED camera do any processing at board-level (I mean does it internally have any image processor) or everything is (post)processed by SDK only? Based on CPU usage, it looks like everything is handled by SDK only but I want to double-check from you.

Thanks again!

Myzhar, 2019-11-14:
Exactly, the CPU usage is not touched, to be more precise is the GPU that is not affected by changing `mat_resize_factor`.
About onboard calculation, the ZED only performs image synchronization, nothing more. All the processing is demanded to the host to take advantage of the CUDA power on GPU.
The CPU only takes care of data transmission and data conversion.

milan-r-shah, 2019-11-14:
Got it. Thank you so much @Myzhar for such quick replies.
Sorry but one more question: I have been using ZED camera mainly with TurtleBot2 (which has Jetson TX2 on it) and checking PonitCloud in RViz in another machine. Both of them are connected to the same wi-fi network. As you pointed, CPU only takes care of data transmission and data conversion. So that's the reason why changing 'mat_resize_factor' affects CPU Usage in Jetson because the amount of data to be transferred to the laptop from Jetson would be affected by that, right?

Myzhar, 2019-11-14:
Exactly, you got the point.
Furthermore the TX2 is not a monster transferring data over Wifi.
I read a few tricks a few time ago about how to improve wifi performance on it, something like the settings you can read here:
https://devtalk.nvidia.com/default/topic/995819/jetson-tx1/jetson-tx1-wifi-speed-and-latency/post/5102535/#5102535

Myzhar, 2019-11-14:
Try also this script:
https://github.com/Rafcin/XavierToolkit/blob/master/enable_faster_internet.sh

milan-r-shah, 2019-11-14:
> Exactly, you got the point.
> Furthermore the TX2 is not a monster transferring data over Wifi.
> I read a few tricks a few time ago about how to improve wifi performance on it, something like the [...]

Thanks a lot @Myzhar for the confirmation.
Oh yes, I have been using the command mentioned in the page you shared. It's very helpful. Thank you for that as well.
In fact, I should have mentioned that command in one of the issues that I closed I think an hour ago.

milan-r-shah, 2019-11-14:
> Try also this script:
> https://github.com/Rafcin/XavierToolkit/blob/master/enable_faster_internet.sh

Sure, thank you @Myzhar . I would post the result here once I try that out.