stereolabs/zed-ros-wrapper #485: Error ZEDmini : Camera not detected,but connected

fatihoezdemir, 2019-10-17:
Hello,
after I connected the ZEDmini and opened ZED Explorer v2.8.3 , the Explorer installed a new firmware and after that  ZED Explorer could not detect the Camera, before that I could see the Stereo Live stream. I think the Firmware is corrupt.
But the camera is connected. I can  see it through **lsusb**
I also cannot update the Camera Firmware through the settings of ZED Explorer because i cannot detect it 
So how can I reinstall a new firmware or delete the corrupt firmware

Myzhar, 2019-10-17:
Hi @fatihoezdemir,
please write an email to support@stereolabs.com for hardware support, our support agents will guide you through the verification process.

closed 2019-10-17 by Myzhar

fatihoezdemir, 2019-10-18:
Hi @Myzhar 
I have contacted the support already yesterday. when will they contact me back. I am a bit in a hurry due to a research project.