stereolabs/zed-ros-wrapper #486: Change map frame_id from launch file

bsubbaraman, 2019-10-18:
Perhaps a basic ROS question, but: I'd like to change the value of the map frame_id from a launch file.  I can change the value from 'map' in the `common.yaml` file, but is there a way I could do this from a launch file?

Myzhar, 2019-10-20:
Hi @bsubbaraman 
you can, it's straightforward.
Take the launch file and search for this part:
```
 <group ns="zed">
        <include file="$(find zed_wrapper)/launch/zed_camera.launch">
            <arg name="svo_file"            value="$(arg svo_file)" />
            <arg name="stream"              value="$(arg stream)" />
            <arg name="node_name"           value="$(arg node_name)" />
            <arg name="camera_model"        value="$(arg camera_model)" />
            <arg name="publish_urdf"        value="$(arg publish_urdf)" />            
        </include>
```
now simply add the line:
```
            <arg name="tracking/map_frame"            value="your_map_frame" />
```

bsubbaraman, 2019-10-20:
Ah, I was just missing the `tracking/` prefix.  Thanks, @Myzhar!

closed 2019-10-20 by bsubbaraman

bsubbaraman, 2019-10-20:
As a note- after testing this, I realized `zed_camera.launch` doesn't accept the map frame (or depth quality) as an argument by default.  As a workaround, I instead added `<param name=zed_node/tracking/map_frame value="my_map_frame" />` to the launch file (instead of as an arg).

Thanks for the help!