stereolabs/zed-ros-wrapper #488: [registerSubscriber] Failed to contact master at [localhost:11311].  Retrying...

peter2527, 2019-10-26:
paltform:Nvidia XAVIER
system:ubuntu18.04
device:zedmini
ros:melodic
when i fellow the document from "https://www.stereolabs.com/docs/ros/",i launch two nodes,error comes out:
terminal1:roslaunch zed_wrapper zedm.launch
terminal2:roslaunch zed_display_rviz display_zedm.launch
and in the terminal showed "[registerSubscriber] Failed to contact master at [localhost:11311].  Retrying..."
and in the terminal shut down and showed"Reason given for shutdown: [new node registered with same name]"
i wonder know this is because of the ros melodic? you allow to use kinetic

Myzhar, 2019-10-26:
Hi @peter2527 
You must not launch two different node.
The `zedm.launch` is used to start a normal ZED-M node, while `display_zedm.launch` starts the ZED node and RVIZ

closed 2020-02-18 by Myzhar

Reschivon, 2021-12-06:
Did you manage to find a solution to this?