stereolabs/zed-ros-wrapper #489: Pulled the newest version of zed wrapper, but cannot disable publish tf

seanxu112, 2019-10-29:
I was 24 commits behind the current master branch. I tried to update zed_wrapper today. After I resolved conflicts, I cannot disable the publish tf feature even though the parameter is set as such. This is proved by the terminal output: 
 * /zed/zed_node/tracking/path_max_count: -1
 * /zed/zed_node/tracking/path_pub_rate: 2.0
 * /zed/zed_node/tracking/pose_topic: pose
 * /zed/zed_node/tracking/publish_map_tf: False
 * /zed/zed_node/tracking/publish_pose_covariance: True
 * /zed/zed_node/tracking/publish_tf: False
 * /zed/zed_node/tracking/spatial_memory: True
 * /zed/zed_node/tracking/two_d_mode: False
 * /zed/zed_node/tracking/world_frame: map
 * /zed/zed_node/video/color_enhancement: True
 * /zed/zed_node/video/left_topic_root: left
 * /zed/zed_node/video/rgb_topic_root: rgb
 * /zed/zed_node/video/right_topic_root: right
 * /zed/zed_node/video/stereo_topic_root: stereo

Since in my older version everything is set to namespace /zed/... instead of /zed/zed_node/... Maybe that is the problem? now I have most topics at /zed/... but with two outliers:
...
...
/zed/right/image_raw_color
/zed/right/image_rect_color
/zed/zed_node/parameter_descriptions
/zed/zed_node/parameter_updates

Here is my zed_no_tf.launch:
```

<?xml version="1.0"?>
<!--
Copyright (c) 2018, STEREOLABS.

All rights reserved.

THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS
"AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT
LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR
A PARTICULAR PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT
OWNER OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL,
SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT
LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; LOSS OF USE,
DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND ON ANY
THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT
(INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE
OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
-->
<launch>
    <arg name="svo_file"              default="" /> <!-- <arg name="svo_file" default="path/to/svo/file.svo"> -->
    <arg name="imu_time_sync"         default="true"/>
    <!-- Camera Model -->
    <arg name="node_name"             default="zed_node" />

    <!-- Camera Model -->
    <arg name="camera_model"          default="zedm" /> <!-- 'zed' or 'zedm' -->

    <arg name="stream"                default="" /> <!-- <arg name="stream" default="<ip_address>:<port>"> -->

    <!-- Publish ZED urdf -->
    <arg name="publish_urdf"          default="true" />

    <arg name="camera_id"             default="-1" />
    <arg name="gpu_id"                default="-1" />

    <group ns="zed">
        <!-- ROS URDF description of the ZED -->
        <group if="$(arg publish_urdf)">
            <param name="zed_description" textfile="$(find zed_wrapper)/urdf/$(arg camera_model).urdf" />
            <node name="zed_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher">
                <remap from="robot_description" to="zed_description" />
            </node>
        </group>

        <node name="$(arg node_name)" pkg="zed_wrapper" type="zed_wrapper_node" output="screen" required="true"><!-- launch-prefix="valgrind" -->
            <rosparam file="$(find zed_wrapper)/params/common.yaml" command="load" />
            <rosparam file="$(find zed_wrapper)/params/$(arg camera_model).yaml" command="load" />

            <!-- SVO file path -->
            <param name="svo_file"          value="$(arg svo_file)" />

            <!-- Remote stream -->
            <param name="stream"            value="$(arg stream)" />

            <!-- Publish ZED urdf -->
            <param name="tracking/publish_tf"        value="false" />
            <param name="tracking/publish_map_tf"    value="false" />

            <!-- Camera ID -->
            <param name="general/zed_id"             value="$(arg camera_id)" />

            <!-- GPU ID -->
            <param name="general/gpu_id"             value="$(arg gpu_id)" />
        </node>
     </group>
</launch>

```

Myzhar, 2019-10-30:
Hi @seanxu112 
I tested the `zed_no_tf` launch file in the latest master branch and it works as expected:
```
[ INFO] [1572436484.778510146]:  * Broadcast odometry TF	-> DISABLED
[ INFO] [1572436484.778990392]:  * Broadcast map pose TF	-> DISABLED
```

Since you have updated the wrapper I suggest you to delete the `build` and `devel` folder s in your catkin workspace and compile the node again.
In this way you will be sure that no old cache values are stored

seanxu112, 2019-10-30:
> Hi @seanxu112
> I tested the `zed_no_tf` launch file in the latest master branch and it works as expected:
>
> ```
> [ INFO] [1572436484.778510146]:  * Broadcast odometry TF	-> DISABLED
> [ INFO] [...]

I have fixed the problem by re-compiling. Just one more quick question. For zedm, is the odom fused with imu? I saw a parameter called imu_fusion, but odom still goes crazy when zedm sees a featureless wall.

Myzhar, 2019-10-31:
With ZEDM you have a completed Inertial Visual Odometry only for the "Pose" topic. The "Odom" topic uses inertial data to estimate the attitude of the camera, but the visual part is dominant and in the case of a featureless wall the result may drift.
The "Pose" topic is instead fixed using spatial memory when a just seen place is detected.

seanxu112, 2019-10-31:
Should I try to merge pose with zed/Odom with robot localization? With imu fusion off?

Myzhar, 2019-10-31:
Yes, you can do it

seanxu112, 2019-10-31:
I am actually cartographer with lidar for mapping and localization. Currently only use ZEDm for odom since the VO is very accurate and with cartographer it localizes very well. The only problem right now is featureless wall. Does Pose requires a map -> odom -> base_link tf? Or it requires the default SLAM that ZED comes with? Is there any way to deal with featureless wall with only ZED to fix odom? I thought about calibrate IMU so that horizontal plane is zero mean, but that requires some time to tune.

Myzhar, 2019-10-31:
Featureless walls are a problem for all the odometry sensor based on vision.
The pose topic is directly generated by the ZED SDK and it natively does not require the Odom frame since it's calculated internally.
The ZED WRAPPER outputs the odom topic just because ROS requires it... 
Take into account one important thing: the pose topic is not an odometry.
The Odometry must be continuous, it cannot have jumps, the pose instead jumps each time a relocation is calculated when a known feature set is detected.

To finally reply to your question: it's not possible to deal with featureless walls only using the ZED.
The universal sensor does not exist, each sensor has weak points:
- A lidar cannot see mirrors and glasses
- A sonar cannot see curtain and soft objects
- A stereo camera requires visual information
- A structured light camera does not work outdoor
- An IMU has bias problems and cannot detect uniform straight motion
- ....
That's why sensor fusion is important to use the strong points of each sensor overcoming the weak points of the others.

closed 2019-11-05 by seanxu112