stereolabs/zed-ros-wrapper #491: reconfigure for mat_resize_factor

eisenWu, 2019-11-04:
Hallo,
sorry for ask the question again, but my question was, could you give me a example how to write the arguments after the service name, in oder to reset the valeu of mat_resize_factor? like:

"set_parameters" "mat_resize_factor 0.2"
"set_parameters" "mat_resize_factor: 0.2"
"set_parameters" "mat_resize_factor float[0.2]"

that are what i have tryed, none of them works.
thanks a lot in advance

eisenWu, 2019-11-04:
i mean call the service "set_parameters"

Myzhar, 2019-11-04:
@eisenWu you must call the methods of the "Dynamic reconfigure" server, you must not use directly the `set_parameters` service:
http://wiki.ros.org/dynamic_reconfigure

eisenWu, 2019-11-04:
@Myzhar i call the `set_parameters` service because i am using the rosbridge to connect ROS and Unity, i can use the `rqt_reconfigure` to reconfigure the `mat_resize_factor`, but i am trying to reconfigure it in Unity side.
thanks for so quick answer:)

Myzhar, 2019-11-04:
Try to see if this thread on ROS Answers can help you:
https://answers.ros.org/question/12276/is-there-a-c-api-for-a-dynamic-reconfigure-client/

eisenWu, 2019-11-04:
thanks a lot

closed 2019-11-04 by eisenWu