stereolabs/zed-ros-wrapper #492: ZED Camera and YD LiDAR do not work together on Jetson TX2

milan-r-shah, 2019-11-04:
I have connected YD LiDAR and ZED Camera to Jetson TX2 which is controlling TurtleBot2. If I launch YD LiDAR first then system stops working and LiDAR also does not start. If I unplug ZED Camera and then try to launch the LiDAR then everything works properly. So, could someone please help me to fix this issue so that I can use both YD LiDAR and ZED Camera together.

Myzhar, 2019-11-04:
Hi @milan-r-shah 
There are two possible causes for your problem:
1) the bandwidth is not enough
2) the port current is not enough

You can try to use an external powered USB3 hub to see if it solves the problem.
If the problem persists then it's a bandwidth problem and I suggest you to move the lidar to the second USB port using a microUSB to USB Type A adapter.

milan-r-shah, 2019-11-05:
Thanks a lot @Myzhar for such a quick response! Sure, I'll try both the options you suggested. 
However, I doubt about either of them. Because I have been facing another issue that counters them.
i.e., let's say in the beginning only YD LiDAR is connected to Jetson TX2 (on TurtleBot2). Also, let's say the robot is running along with YD LiDAR. No problem, so far! 
So, if I connect the ZED Camera now, when the robot is already ON (as well as YD LiDAR) and then I try to launch ZED ROS wrapper then, this time, it works. In fact, the robot doesn't stop working!!

_It did affect the tf tree_ but the robot didn't stop working unlike in the previous case, I mentioned above.

Myzhar, 2019-11-05:
In that case I'm pretty sure it's a current problem since starting both the sensors at the same time requires more current and they fail because the port itself cannot source enough.

milan-r-shah, 2019-11-05:
This makes sense and definitely I would try giving more current to the LiDAR but just to clarify, I'm not starting both the sensors at the same time. In the beginning, both the sensors are just connected, not started. In that case, if I first try starting YD LiDAR then entire system i.e. the robot+LiDAR get affected and the robot stops working.

milan-r-shah, 2019-11-14:
I was able to solve the issue through both the methods you suggested i.e. 1. connecting the lidar to different USB port i.e. micro-USB port and the camera to main USB port. 2. Using an external powered USB hub.
Thanks a lot @Myzhar again. I really appreciate your quick suggestions/help.

closed 2019-11-14 by milan-r-shah