stereolabs/zed-ros-wrapper #493: Can I get synchronized left and right image?

prarobo, 2019-11-12:
When I subscribe to left and right camera images, I see that the timestamps between images do not match. Is it possible to get synchronized left and right image streams?

Myzhar, 2019-11-12:
Hi @prarobo 
left and right images are published in a synchronous way. 
To be sure to receive synchronized images in another node you must use the right ROS tool:
http://wiki.ros.org/message_filters#Time_Synchronizer

Myzhar, 2020-03-23:
We added a nodelet to be used as an example to get synchronized topics.
See https://github.com/stereolabs/zed-ros-wrapper/tree/master/zed_nodelets/src/rgbd_sensors_sync_nodelet
and the relative documentation:
https://www.stereolabs.com/docs/ros/zed_nodelets/

closed 2020-03-23 by Myzhar