stereolabs/zed-ros-wrapper #494: Using openni depth mode results in unrecognized image encoding

bsubbaraman, 2019-11-17:
I'm trying to use the openni depth mode with my zed. 

First, as a side note, when changing the value in `common.yaml`, the mode only changes if the value is set to `true`; when set to `1` as the inline comment says, `openni_depth_mode` remains disabled. Is this expected?

Second, when interfacing with cv_bridge, trying to use the line
 `cv_ptr = cv_bridge::toCvCopy(_depth_frame_from_sensor)` 
results in the error
`cv_bridge exception: Unrecognized image encoding []`

This line/the rest of my code works fine when `openni_depth_mode` is false. If I echo `depth/depth_raw_registered/encoding`, I see that it does indeed change from `32FC1` to `mono16`.

 Am I using openni_depth_mode incorrectly? Any help is greatly appreciated!

Myzhar, 2019-11-18:
Hi @bsubbaraman 
maybe that `cv_bridge` does not recognize `mono16` as image format.

Can you please try to replace `MONO16` with `TYPE_16UC1` in the following line:
https://github.com/stereolabs/zed-ros-wrapper/blob/384347e29c9702725f8943fe0e7439373336244a/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp#L1423
then recompile and see if it works.

Thank you
Walter

bsubbaraman, 2019-11-18:
Thanks for the response and idea! Unfortunately, after making the change and recompiling, I get the same error. In fact, even using `TYPE_32FC1` gives me the same error, even though that is the encoding that is used when `openni_depth_mode` is false.

Any other things to try? `cv_bridge` is supposed to print the name of the unrecognized encoding inside of the brackets `[]`, which leads me to believe that the message is not being made/formatted correctly (my `cv_bridge` properly prints out `32FC1` when I check the encoding not using `openni_depth_mode`).

bsubbaraman, 2019-11-18:
Additionally, if I simply change the following line, which is the method called to publish messages when `openni_depth_mode` is false:

https://github.com/stereolabs/zed-ros-wrapper/blob/384347e29c9702725f8943fe0e7439373336244a/zed_wrapper/src/tools/src/sl_tools.cpp#L196

to `TYPE_16UC1`, my code runs fine and `depth/depth_raw_registered/encoding` properly echoes `16UC1`. (This doesn't do any conversion so its not a fix, but confirms that my `cv_bridge` can accept the proper encoding).

bsubbaraman, 2019-11-26:
Apologies for the delay- following up to note that I'll report back when I'm able to give this a shot.

Myzhar, 2020-03-23:
Closed for inactivity

closed 2020-03-23 by Myzhar