stereolabs/zed-ros-wrapper #495: Launching zed-ros-wrapper freezes Jetson TX2

milan-r-shah, 2019-11-19:
Hi,
I have been using a ZED Camera with Jetson TX2 (Ubuntu 16 & ROS Kinetic) and generally, I use my local machine (also, Ubuntu 16 & ROS Kinetic) to connect to Jetson through SSH as both of them are in the same Wi-Fi network. 
But this time, whenever I'm trying to launch zed-ros-wrapper, it freezes Jetson. Neither I'm able to do anything in other logged-in terminals nor am I able to ping Jetson from my local machine. This happened one or two times before as well but then it started working normally on the next day.
(Note: I have already turned off wireless power save mode in Jetson TX2)
Could you please help me to fix this issue permanently?

Thanks a lot in advance!

Myzhar, 2019-11-19:
Hi @milan-r-shah 
it seems a problem with your TX2. 
Please check that full power is enabled (jetson_clocks started).
Is the fan running at full speed and the heatsink correctly installed?

I suggest you to install this package:
https://github.com/rbonghi/jetson_stats
it allows you to verify that everything is correctly working on your TX2, mainly that the temperatures are not too high.

milan-r-shah, 2019-11-25:
Hi @Myzhar 
Thanks for all the suggestions. Currently, in my Jetson, full power is already enabled, the fan is running at full speed, the heatsink correctly installed, and the temperature is also normal. But still, I'm having the same issue.
Another weird thing I'm observing is when I connect Jetson to Monitor, connect keyboard & mouse and then if I launch the wrapper from Jetson itself (I mean without using SSH from another machine), I don't face the problem, I mentioned earlier.
Any other suggestions would greatly be appreciated. Thanks in advance!

Myzhar, 2019-11-25:
Ok, you added a new variable to the equation: HDMI.
One of the problem with Jetson and CUDA is that the graphic environment must be started to be able to use CUDA and the ZED SDK requires it to work.

You can read more about the problem here:
https://www.jetsonhacks.com/2016/05/03/jetson-allow-graphics-without-hdmi/

milan-r-shah, 2019-11-27:
Hi @Myzhar,
Thanks for sharing another link. I checked that out but my /etc/X11/xorg.conf file has already been up to date i.e. _Option "AllowEmptyInitialConfiguration"_ is already _"true"_.

However, I found another problem (and because of that, probably, I would update the title of this issue #495):
I connected everything to the Jetson i.e. ZED camera, Lidar, monitor, keyboard, etc. Now, whenever I launched other ROS packages, there wasn't any problem. But, as soon as, I launched zed-ros-wrapper, the jetson got disconnected from the wi-fi. Apart from wi-fi, everything else was working fine. But because of that, the Jetson was not able to access the internet. So, eventually, it can't be accessed through SSH.

My current guess is this is the main issue that has been stopping me to use SSH from another machine to access Jetson. Last time, I was launching the zed-ros-wrapper through SSH and probably, it would have affected wi-fi and that's why the terminal would have become inactive.

So, how does zed-ros-wrapper affect the wi-fi? Because, as soon as, I stop the wrapper, the jetson gets connected back to the wi-fi. Thanks!

Myzhar, 2019-11-27:
Ok, this seems to be a bandwidth problem. You are saturating the wifi and so the SSH console freezes waiting for data.

This issue can help you:
https://github.com/stereolabs/zed-ros-wrapper/issues/469

milan-r-shah, 2019-11-27:
@Myzhar 
Thanks for the quick response!

I tried both methods:
1. Used ROS topic_tools/throttle and reduced the "/zed/zed_node/point_cloud/cloud_registered" to 5. But it didn't help. I'm confused that should I throttle any other topics as well?

2. Reduced the point_cloud_freq and frame_rate to 5, general/resolution to 3 (the lowest), mat_resize_factor to 0.5, depth_stabilization to 0, mapping/resolution to 2 (LOW). But still, facing the same issue. As soon as I'm launching zed-ros-wrapper, wi-fi on the Jetson stops working so I can't use SSH.

Is there anything else that I should/can check? Please let me know. Thanks in advance!

Myzhar, 2019-11-27:
Have you tried if using a cabled network you have the same issue?

milan-r-shah, 2019-11-27:
Not yet. But based on the requirements, the robot (i.e. Jetson) needs to be connected through wi-fi.

Myzhar, 2019-11-27:
It's only to exclude that it's a problem with wifi and to confirm that it's a bandwidth issue

milan-r-shah, 2019-11-27:
Hi @Myzhar, 
Thanks for the suggestion of using a cabled network. I just tried connecting the Jetson to a local machine using a cross-over ethernet cable. In that configuration, I didn't face any problems that I have been facing while using a wi-fi connection. 
So, now, could you please suggest any ways through which I can run the zed-ros-wrapper through the wi-fi?
Thanks a lot in advance!

milan-r-shah, 2019-12-16:
Hi @Myzhar ,
I'm still facing this issue so if you have any further suggestions then please let me know.
Thanks in advance!

federicatrozzi, 2020-02-18:
Hi @Myzhar  i've got the same problem! any updates?
Maybe i can add another variable in this equation: in my case after the upgrade to zed sdk 3.0 i'm able to use zed-ros-wrapper and visualize in rviz some images (but in a very laggy way, streaming is ridiculously too slow). 
But after few moments (that can be minutes or hours) my connection to wifi drops down and i have to reboot manually my TX2 board.
i'm using ROS Melodic + ubuntu 18.04+ jetpack 4.2
Thank you

Myzhar, 2020-02-18:
The problem with wifi is that the bandwidth is very limited and the reliability of the connection depends on too many variables.

The available bandwidth is regulated by the wifi driver and it is reduced if connection is not strong.
To check the real wifi bandwidth you can use the CLI command:
`$ iwconfig <connection_name>`
you can get the value for <connection_name> using the command:
`$ ifconfig`
At this point the field `Bit Rate` gives you the exact value of the available bandwidth for your connection.

The CLI command
`$ rostopic bw <topic_name>`
is now helpful to get a statistic about the bandwidth required by each topic you are subscribing. 
You can run it on the same machine where the ZED wrapper is running to not be limited by wifi while getting this information.

What you must do now is quiet straightforward: use the command `rostopic bw` for each topic you want to transmit over wifi and sum al the bandwidths required by each topic.
If the sum is major of the available bandwidth then the wifi will be inevitably congested.

The only solution is to tune the data sent on the network finding the best solution that fits your requirements.

You can evaluate many possible solutions from the following list, also using a combination of them:
1) use 5Ghz WIFI instead of 2.4Ghz to increase the bandwidth, but reducing the wifi range
2) subscribe to less topics over wifi
3) reduce data resolution in ZED Wrapper
4) reduce data FPS in ZED Wrapper
5) use `throttle` in [`topic_tools`](http://wiki.ros.org/topic_tools/throttle)
6) use the new [`video/img_resample_factor`](http://wiki.ros.org/zed-ros-wrapper#Video_Parameters) and/or [`depth/depth_resample_factor`](http://wiki.ros.org/zed-ros-wrapper#Depth_Parameters) parameters in the ZED wrapper to reduce the amount of data for images and depth information

federicatrozzi, 2020-02-18:
Hi @Myzhar ,
thank you for the fast reply. Unfortunately nothing to do with any combination you suggested me. 
The Bit Rate of my wifi network is 40,5M/s and i'm using 5Ghz frequency. 
The CLI command `rostopic bw /zed/zed_node/rgb/image_rect_color` gave me the average of 1.26kb/s using the resolution of VGA@15 fps and a video/img_resample_factor of 0.1 (so the image transmitted now is  67x37 pixels.
Even this attempts, the command `rostopic hz /zed/zed_node/rgb_raw/image_raw_color`
give me 
`no new messages
no new messages
no new messages
no new messages
average rate: 0.124
	min: 8.034s max: 8.034s std dev: 0.00000s window: 2
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
`
(and also using throttle in topic_tools i had same results)
i ran this command both on my laptop and my TX2 so, can i suppose that is not a streaming problem but something else that not depends on wifi? thank you

federicatrozzi, 2020-02-18:
And the only topics i'm publishing are the zed-ros-wrapper ones :)

Myzhar, 2020-02-18:
That's weird. I expect what you wrote about `rostopic hz /zed/zed_node/rgb_raw/image_raw_color` if you run the command on the remote device, not locally.
Where is the `roscore` running? Locally to ZED node or on the remote machine?

Myzhar, 2020-02-18:
I suggest also to run the CLI command `roswtf` both on local and remote machine to understand if there is something wrong with your ROS configuration

federicatrozzi, 2020-02-18:
> That's weird. I expect what you wrote about `rostopic hz /zed/zed_node/rgb_raw/image_raw_color` if you run the command on the remote device, not locally.
> Where is the `roscore` running? Locally to ZED node or on the remote machine?

roscore is running on the remote machine where also the ZED is. Just to explain better: i have a remote racecar with on it a ZED and a TX2 and i ssh it from my local laptop to control it.

Myzhar, 2020-02-18:
It's a classical configuration that we tested a lot for robotics applications.
Normally the communication issues are related to `jetson_clocks` not started and Jetson boards not running at maximum performances or to wifi bandwidth saturated.
Just to exclude any other variable, have you tried a directed cabled connection to test if in this way everything works or not?
This is usually the first test I suggest to be performed to exclude many other possible problems

federicatrozzi, 2020-02-19:
Unfortunately `roswtf` gave me only useless warnings. BUT! i found another clue and i don't know how to fix it. 
I have a "start_sensors.launch" custom file that launches nodes of sensors + the zed one. 
If i set parameter `<arg name="publish_urdf"        value="false" />` the streaming of images topics have the `average rate: 0.124` but if i set it `true` i have the correct average rate (15fps) and the streaming goes fine. 
The problem is that if i set it `true` my localization algorithm does not work (i think is related to a static tf transform).

In fact in my start_sensors. launch i have:

`	
	<param name="robot_description" command="$(find xacro)/xacro --inorder $(arg model)" />
	<node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" output="screen" />`

and in the zed_camera.launch if i set `<arg name="publish_urdf"        value="true" />` i have:

`        <node name="zed_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" output="screen" required="true">
            <remap from="robot_description" to="zed_description" />
        </node>`

i'm investigating on it, but if you have it i appreciate a tip :) 

Thank you for your time

Myzhar, 2020-02-19:
Hi @federicatrozzi 
this seems to be a new problem. Can you please create a new issue so we can track it?

federicatrozzi, 2020-02-19:
Yes sure!

Myzhar, 2020-03-23:
The new features added to the ZED Wrapper can be used to reduce the bandwitch usage.
See the parameters:
- `pub_frame_rate`
- `video/img_downsample_factor`
- `depth/depth_downsample_factor`
The ROS wiki has been updated with all the new information: http://wiki.ros.org/zed-ros-wrapper

closed 2020-03-23 by Myzhar