stereolabs/zed-ros-wrapper #496: Non-static transform from base_link to camera

MikeWrockAIS, 2019-11-22:
I noticed that even though you can set the frame in which to publish the ZED's odometry information, the current TF is not used. For example: if the camera is on a moving arm, when the arm moves the camera will publish odometry as if base_link moved, ignoring the current base_link to camera transform. Is there any way to tell the zed-wrapper the camera moves relative to base link during operation?

Myzhar, 2019-11-22:
Hi @MikeWrockAIS 
the default URDF is designed for robots with the camera statically linked to `base_link`, but you can change it according to your configuration.

In your case you can simply change the type of joint between `zed_camera_center` and `base_link` from `fixed` to `floating`
https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/urdf/zed.urdf#L25

In this way the `robot_state_publisher` will not publish the relative TF, but you should do it in one of your nodes, otherwise the TF of the ZED will result as broken.

MikeWrockAIS, 2019-11-22:
Does it have to be the `zed_camera_center` to `base_link` joint that is floating or can it be one joint along the TF tree such as:
`base_link` -> `rotating_arm` (`floating`)
`rotating_arm` -> `zed_camera_center` (`fixed`)

Myzhar, 2019-11-25:
It depends on your configuration. You should modify the ZED URDF according to how the camera is linked on the robot. The important thing is that the chain `zed_camera_center` -> .... -> `base_link` is not broken and the ROS TF can always calculate the `zed_camera_center` -> `base_link` transform

MikeWrockAIS, 2019-11-25:
It seems like once a valid transform from camera to base link is acquired [mCamera2BaseTransfValid is set to true](https://github.com/stereolabs/zed-ros-wrapper/blob/24f00b647b413c913c4c5a06ce512c1d5136bd9d/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp#L890). Since [the transform is only updated if mCamera2BaseTransfValid is false](https://github.com/stereolabs/zed-ros-wrapper/blob/24f00b647b413c913c4c5a06ce512c1d5136bd9d/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp#L993-L995), I don't see how providing an up to date TF tree helps. Perhaps I should set mCamera2BaseTransfValid to false or simply remove the if statement and call  getCamera2BaseTransform() every time?

Myzhar, 2019-11-25:
Yes, in your case you must update it at every cycle so you must remove the check.
We will add a parameter to discriminate between static and not static transform.

MikeWrockAIS, 2019-11-25:
awesome, thanks a lot

Myzhar, 2019-11-25:
I keep the issue open as reminder for the feature to be added

tugbakara, 2022-08-05:
I have a smilar issue, tomorrow I will open a new issue and add the details.