stereolabs/zed-ros-wrapper #498: get error when compiling zed-ros-wrapper using catkin_ws

fzrn23, 2019-11-30:
SDK version :2.5.1 jeston TX2
Ubuntu 16.04 & ROS Kinetic

I have done the following steps:
cd ~/catkin_ws/src
git clone https://github.com/stereolabs/zed-ros-wrapper.git
cd ../

but when I used  catkin_make  I received this :

zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/build.make:86: recipe for target 'zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/src/nodelet/src/zed_wrapper_nodelet.cpp.o' failed
make[2]: *** [zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/src/nodelet/src/zed_wrapper_nodelet.cpp.o] Error 1
CMakeFiles/Makefile2:3285: recipe for target 'zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/all' failed
make[1]: *** [zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j6 -l6" failed

What should I do to solve this problem?

Myzhar, 2019-11-30:
Hi @fzrn23
have you copied and pasted the full error message? 
I can't see any explanation about what's the cause of the compiling error.  

However you are using a very old version of the SDK. I suggest you to download the latest SDK v2.8.4: https://www.stereolabs.com/developers/release/

fzrn23, 2019-12-01:
@Myzhar 
I updated the SDK and this problem was solved.
Thanks for your help.

Myzhar, 2019-12-01:
You are welcome 👍

closed 2019-12-01 by Myzhar