stereolabs/zed-ros-wrapper #499: Zed wrapper issue when using <machine> tag

radisc, 2019-12-03:
Hello,

i'm trying to launch the zed ros wrapper from a launch file in another machine using the [ machine tag](http://wiki.ros.org/roslaunch/XML/machine).

I tested this method it with another generic camera driver ( e.g. video_stream_opencv with the zed specifing /dev/video0 ) and it works like a charm.

However, if i do the same with the zed ros wrapper, i get awful performance ( not even a frame every 60 seconds, in fact i think is totally hang ).

Do you have any idea of what might be the problem?

Thanks

<launch>

	<machine name="dell" address="10.42.0.1" env-loader="/home/nicola/Workspace/Kinetic/catkin_ws/devel/env.sh" user="nicola" default="true"/>

	
	<node machine="dell" pkg="roi_to_bounding_box" name="people_detector" type="people_detector" output="screen" required="true">
		
		
		<param name="scaling_factor" value="1.0" />
		<param name="min_confidence" value="-0.10"/>
		<param name="AA"             value="3.0"/>
        <param name="BB"             value="2.0"/> 
    </node> 

<group>

         Define the remote machine 
	<machine name="jesto-nano" address="10.42.0.221" env-loader="/home/nvidia/en_loader.sh" user="nvidia" default="never"/>

        
	<node machine="jesto-nano" name="stream_opencv" pkg="video_stream_opencv" type="video_stream" output="screen" >
	</node>
        
	<node machine="jesto-nano" name="zed_node" pkg="zed_wrapper" type="zed_wrapper_node" output="screen" > 
        <rosparam file="$(find zed_wrapper)/params/common.yaml" command="load" />
        <rosparam file="$(find zed_wrapper)/params/zed.yaml" command="load" />
    </node>

</group>

with this workaround i can launch a script that open an ssh connection and executes a command on the remote machine, it would be "fine" but the nodes do not get shut down when i Ctrl+C on the main machine

	<node pkg="roi_to_bounding_box" type="remote_launch.sh" name="remote_launch" output="log"  /> 
</launch>

jesto-nano is the actual name of the computer, i put a typo when i set it up...

Myzhar, 2019-12-12:
Hi @radisc 
you are saturating the bandwidth of your connection, that's why you are getting low performances.

There are a lot of discussions about this problem:
https://github.com/stereolabs/zed-ros-wrapper/issues/495
https://github.com/stereolabs/zed-ros-wrapper/issues/473
https://github.com/stereolabs/zed-ros-wrapper/issues/469

radisc, 2019-12-12:
Hi @Myzhar , 

thank you for the reply, but i checked, is not a bandwidth problem, look:
[image]
This is the zed_wrapper_node launched remotely

[image]
And this is video_stream_opencv, as you can see i'm directly connected over a gigabit connection.

Myzhar, 2019-12-13:
I see, I try to understand your problem testing it by my side.
Please provide the solution if you find it meanwhile

Myzhar, 2020-03-23:
@radisc have you found a solution for this issue?

Myzhar, 2020-03-26:
Closed for inactivity

closed 2020-03-26 by Myzhar

jgbreen, 2021-09-25:
Hi @Myzhar,

I have been trying to solve a similar issue to this issue (#499) as well as the other issues (#495, #473 and #469) where I can no longer connect to the wifi hotspot that is setup on my Jetson TX1 when I run the zed wrapper node for the zed2i. I dont believe it is a bandwidth problem as when I run the node my network status is fine until it drops out (similar to the figure in this thread).

I was wondering if you have found a solution to this.

Thanks in advanced!

Myzhar, 2021-09-25:
Hi @jgbreen 
please do not reply to a closed ticket, always open a new one and put there a link to all the closed issues that are interesting for your problem.