stereolabs/zed-ros-wrapper #500: catkin_make error

willzoe, 2019-12-12:
CUDA: 9.0
SDK version: 2.8.3
Ubuntu 16.04
ROS Kinetic

I have done the following steps on my PC:

> $ mkdir -p ~/zed_nav_original/src
$ cd ~/zed_nav_original/src/
$ git clone -b nav_tutorial https://github.com/stereolabs/zed-ros-wrapper.git
$ cd ..
$ catkin_make

but when I used catkin_make I received this :

```
Base path: /home/scuimg/zed_nav_original
Source space: /home/scuimg/zed_nav_original/src
Build space: /home/scuimg/zed_nav_original/build
Devel space: /home/scuimg/zed_nav_original/devel
Install space: /home/scuimg/zed_nav_original/install
Creating symlink "/home/scuimg/zed_nav_original/src/CMakeLists.txt" pointing to "/opt/ros/kinetic/share/catkin/cmake/toplevel.cmake"
####
#### Running command: "cmake /home/scuimg/zed_nav_original/src -DCATKIN_DEVEL_PREFIX=/home/scuimg/zed_nav_original/devel -DCMAKE_INSTALL_PREFIX=/home/scuimg/zed_nav_original/install -G Unix Makefiles" in "/home/scuimg/zed_nav_original/build"
####
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/scuimg/zed_nav_original/devel
-- Using CMAKE_PREFIX_PATH: /home/scuimg/turtlebot_original/devel;/home/scuimg/kobuki/devel;/home/scuimg/rocon/devel;/opt/ros/kinetic
-- This workspace overlays: /home/scuimg/turtlebot_original/devel;/home/scuimg/kobuki/devel;/home/scuimg/rocon/devel;/opt/ros/kinetic
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/scuimg/zed_nav_original/build/test_results
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.18
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 5 packages in topological order:
-- ~~  - zed_display_rviz
-- ~~  - zed_navigation_tutorial
-- ~~  - zed_nodelet_example
-- ~~  - zed_rtabmap_example
[... 470 lines cut ...]
       ROSCONSOLE_PRINT_STREAM_AT_LOCATION(args); \
       ^
/opt/ros/kinetic/include/ros/console.h:568:43: note: in expansion of macro ‘ROS_LOG_STREAM_COND’
 #define ROS_LOG_STREAM(level, name, args) ROS_LOG_STREAM_COND(true, level, name, args)
                                           ^
/opt/ros/kinetic/include/rosconsole/macros_generated.h:113:43: note: in expansion of macro ‘ROS_LOG_STREAM’
 #define ROS_INFO_STREAM_NAMED(name, args) ROS_LOG_STREAM(::ros::console::levels::Info, std::string(ROSCONSOLE_NAME_PREFIX) + "." + name, args)
                                           ^
/opt/ros/kinetic/include/nodelet/nodelet.h:60:34: note: in expansion of macro ‘ROS_INFO_STREAM_NAMED’
 #define NODELET_INFO_STREAM(...) ROS_INFO_STREAM_NAMED(getName(), __VA_ARGS__)
                                  ^
/home/scuimg/zed_nav_original/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:771:9: note: in expansion of macro ‘NODELET_INFO_STREAM’
         NODELET_INFO_STREAM("Floor Alignment : " << trackParams.enable_floor_alignment);
         ^
zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/build.make:110: recipe for target 'zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/src/nodelet/src/zed_terrain_mapping.cpp.o' failed
make[2]: *** [zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/src/nodelet/src/zed_terrain_mapping.cpp.o] Error 1
zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/build.make:86: recipe for target 'zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/src/nodelet/src/zed_wrapper_nodelet.cpp.o' failed
make[2]: *** [zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/src/nodelet/src/zed_wrapper_nodelet.cpp.o] Error 1
CMakeFiles/Makefile2:784: recipe for target 'zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/all' failed
make[1]: *** [zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
```
What should I do to solve this problem?

Myzhar, 2019-12-12:
Hi @willzoe
you have checked out a wrong branch that we are using for testing.
Only the `master` branch can be used with the ZED SDK.

Try to go to the `zed-ros-wrapper` folder and run the command
`$ git checkout master` 
before compiling

willzoe, 2019-12-12:
@Myzhar 
Can `zed_navigation_tutorial` in this branch be used in the `master` branch?

I want to use a ZED stereo camera instead of the kinect RGBD camera for the turtlebot2 to implement mapping and autonomous navigation. But I don't know how to do it.

Myzhar, 2019-12-12:
You cannot because it uses information that are not available and will generate errors in your case.

To integrate the ZED in the Turtlebot you must replicate what is done for Kinect or other 3D sensors, replacing their node with the ZED node

willzoe, 2019-12-12:
@Myzhar Thanks for your patience
I will try to replace other 3D sensors nodes with zed nodes in `turtlebot_navigation` package.

closed 2019-12-12 by willzoe