stereolabs/zed-ros-wrapper #502: imu and camera base line value

vigneshsara, 2019-12-30:
hi author,
i had done the imu calibration using kalibr on zed-mini and obtain result almost same your example value (https://support.stereolabs.com/hc/en-us/articles/360012749113-How-can-I-use-Kalibr-with-the-ZED-Mini-camera-in-ROS-).
i had some doubt on the give value .
camera base line =63mm 
camera0 to imu  base line=  -0.06287466260313691(almost 63mm)
The above value denote the both camera and imu sensor are in same position. This is correct result ???

Myzhar, 2020-01-13:
Hi @vigneshsara 
the IMU sensor is indeed very near to the camera sensor. You can get the transform matrix just starting the ZED wrapper node and looking at the log output on the console:

this is for example the transform matrix for one of the ZED2 cameras:
```
$ roslaunch zed_wrapper zed2.launch

[...]
[ INFO] [1578904344.511432279]:  ...  ZED ready
Camera-IMU Transform: 
 7FFC07B1BE60
0.999626 0.025546 0.009792 -0.002000
-0.025588 0.999664 0.004221 -0.023000
-0.009681 -0.004470 0.999943 -0.002000
0.000000 0.000000 0.000000 1.000000
[...]
```

closed 2020-02-18 by Myzhar

MickyasTA, 2025-11-25:
Hello, I used this code to get the transformation matrix, but I was not sure if this "sensors_configuration.camera_imu_transform" gives me the IMU-to-camera transform or the camera-to-IMU transform. Could you please clarify it for me? 

And also, what is the coordinate system of the Zed2 IMU and both of the cameras, left and right? 

```

import os
import csv
import time
import argparse
import datetime
from pathlib import Path
from threading import Thread, Lock
import copy
import json
import numpy as np
import cv2
import pyzed.sl as sl
import os, sys
sys.path.append(os.path.join(os.path.dirname(__file__), "mocap/mocap_ros"))
from scipy.spatial.transform import Rotation as R

zed = sl.Camera()

init = sl.InitParameters()
init.camera_resolution = sl.RESOLUTION.HD1080
init.camera_fps = 30
init.depth_mode = getattr(sl.DEPTH_MODE, "NEURAL")
init.coordinate_units = sl.UNIT.METER
init.coordinate_system = sl.COORDINATE_SYSTEM.IMAGE
zed.open(init)

info = zed.get_camera_information()
cam_imu_transf = info.sensors_configuration.camera_imu_transform  # IMU to Left Camera transform
save_path = Path("zed_imu_transform.json")
cam_imu_list = cam_imu_transf.m.tolist()  # .m gives the 4x4 matrix

with open(save_path, "w") as f:
    json.dump({
        "Transformation Matrix": {
            "cam_imu_transf": cam_imu_list
        }
    }, f, indent=4)
print(cam_imu_transf)
```