stereolabs/zed-ros-wrapper #503: Pointcloud Tilt

Mizbauddin, 2020-01-02:
Hi, I am using tilting the ZED camera for my use case and the point cloud gets tilted too. This is how the point cloud currently looks.

[image]

I did read in other issues about changing the URDF file. I am not sure what parameters need to be changed to get the orientation relative to the base_frame of the robot. 

Thank you.

Myzhar, 2020-01-03:
Hi @Mizbauddin 
you are right, the URDF must be changed to reflect the tilting. 
You should add `3.14` to all the joint rotations around the X axis:
```
<!-- Camera Center -->
    <joint name="zed_camera_center_joint" type="fixed">
        <parent link="base_link"/>
        <child link="zed_camera_center"/>
        <origin xyz="0 0 0.0" rpy="3.141592654 0 0" />
    </joint>

    <link name="zed_camera_center">
        <visual>
            <origin xyz="0 0 0" rpy="0 0 0"/>
            <geometry>
                <mesh filename="package://zed_wrapper/urdf/models/ZED.stl" />
            </geometry>
            <material name="light_grey">
                <color rgba="0.8 0.8 0.8 0.8"/>
            </material>
        </visual>
    </link>

<!-- Left Camera -->

    <joint name="zed_left_camera_joint" type="fixed">
        <parent link="zed_camera_center"/>
        <child link="zed_left_camera_frame"/>
        <origin xyz="0 0.06 0" rpy="3.141592654 0 0" />
    </joint>

    <link name="zed_left_camera_frame" />

    <joint name="zed_left_camera_optical_joint" type="fixed">
        <origin xyz="0 0 0" rpy="1.57079632679 0.0 1.57079632679"/>
        <parent link="zed_left_camera_frame"/>
        <child link="zed_left_camera_optical_frame"/>
    </joint>

    <link name="zed_left_camera_optical_frame"/>

<!-- Right Camera -->

    <joint name="zed_right_camera_joint" type="fixed">
        <parent link="zed_camera_center"/>
        <child link="zed_right_camera_frame"/>
        <origin xyz="0 -0.06 0" rpy="3.141592654 0 0" />
    </joint>

    <link name="zed_right_camera_frame" />

    <joint name="zed_right_camera_optical_joint" type="fixed">
        <origin xyz="0 0 0" rpy="1.57079632679 0.0 1.57079632679"/>
        <parent link="zed_right_camera_frame"/>
        <child link="zed_right_camera_optical_frame"/>
    </joint>

    <link name="zed_right_camera_optical_frame"/>```

Mizbauddin, 2020-01-07:
Hi @Myzhar Thank you so much for getting back. I am using zed mini and I am tilting the camera by 45 degrees forward.  Do I still need to add 3.14 to all the joint rotations or instead add 0.785398 (π / 4) to the joint rotations? 

Also, the URDF file has an IMU joint. Do I also need to add 3.14 to that? Thank you.

Myzhar, 2020-03-23:
The new "xacro" feature should solve this issue. See the new launch files

closed 2020-03-23 by Myzhar