stereolabs/zed-ros-wrapper #504: ResourceNotFound: zed_wrapper

willzoe, 2020-01-05:
I have five catkin workspaces, `rocon`, `kobuki`, `turtlebot`, `zed`, and `myrobot`. In `myrobot`, I created three packages that depend on  `zed_wrapper`.

Today, when I run the launch file of one of the packages in `myrobot`, the system reminds me that `zed_wrapper` cannot be found. So I checked the environment variable `ROS_PACKAGE_PATH` and found that it does not contain the workspace `zed` where `zed_wrapper` is located.

I added `zed` to `ROS_PACKAGE_PATH` by executing the following command：
```
source /home/nvidia/zed/devel/setup.bash
```
After this `zed_wrapper` can be found, but the `/home/nvidia/myrobot/src` originally contained in `ROS_PACKAGE_PATH` is gone, that is, I cannot execute the launch file of the packages in `myrobot`.

If I execute the following command to add `myrobot` to `ROS_PACKAGE_PATH`, then `/home/nvidia/zed/src` will disappear from `ROS_PACKAGE_PATH` again, so `zed_wrapper` will not be found again.
```
source /home/nvidia/myrobot/devel/setup.bash
```
Why is this happening?

Myzhar, 2020-01-05:
Hi @willzoe 
You are doing only one wrong thing: you are separating the catkin workspace.
Your catkin workspace must contain ALL the packages of your robot in the `src` folder.
For example:
```
/home/nvidia/catkin_ws/src/
|
- zed-ros-wrapper
- myrobot
- pkg1
- pkg2
...
```
at this point you must source the whole catkin workspace:
`source /home/nvidia/catkin_ws/devel/setup.bash`

You can read more here:
http://wiki.ros.org/catkin/Tutorials/using_a_workspace

willzoe, 2020-01-05:
It seems to be because I inappropriately created a package for navigation in `myrobot`. When I delete this package, I can find the packages in each worksapce normally.

closed 2020-01-05 by willzoe