stereolabs/zed-ros-wrapper #506: catkin_make error

Scarf195, 2020-01-15:
ZED SDK: 2.8.5
CUDA 10.2
Distro ROS: Melodic
Ubuntu 18.04

Hi all, I have done the following steps:
cd ~/catkin_ws/src
git clone https://github.com/stereolabs/zed-ros-wrapper.git
cd ../
but when I do catkin_make -DCMAKE_BUILD_TYPE=Release  I received this:

...
...
[ 91%] Built target zed_wrapper_generate_messages_cpp
[ 91%] Built target zed_wrapper_generate_messages
make[2]: ***  Nessuna regola per generare l'obiettivo "/usr/local/cuda-10.2/lib64/libcublas.so", necessario per "/home/scarf/zed_ws/devel/lib/libZEDWrapper.so".  Arresto.
CMakeFiles/Makefile2:2537: recipe for target 'zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/all' failed
make[1]: *** [zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

What have I do to solve this problem?

Scarf195, 2020-01-15:
This is in english version:

~/zed_ws$ catkin_make -DCMAKE_BUILD_TYPE=Release
Base path: /home/scarf/zed_ws
Source space: /home/scarf/zed_ws/src
Build space: /home/scarf/zed_ws/build
Devel space: /home/scarf/zed_ws/devel
Install space: /home/scarf/zed_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/scarf/zed_ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/scarf/zed_ws/build"
####
[  1%] Built target zed_wrapper_gencfg
[ 11%] Built target zed_depth_sub
[ 11%] Built target zed_video_sub
[ 11%] Built target zed_tracking_sub
[ 11%] Built target _zed_wrapper_generate_messages_check_deps_set_pose
[ 11%] Built target _zed_wrapper_generate_messages_check_deps_start_svo_recording
[ 11%] Built target _zed_wrapper_generate_messages_check_deps_toggle_led
[ 11%] Built target _zed_wrapper_generate_messages_check_deps_reset_tracking
[ 11%] Built target _zed_wrapper_generate_messages_check_deps_reset_odometry
[ 11%] Built target _zed_wrapper_generate_messages_check_deps_set_led_status
[ 11%] Built target _zed_wrapper_generate_messages_check_deps_start_remote_stream
[ 11%] Built target _zed_wrapper_generate_messages_check_deps_stop_svo_recording
[ 11%] Built target _zed_wrapper_generate_messages_check_deps_stop_remote_stream
[ 72%] Built target zed_wrapper_generate_messages_nodejs
[ 74%] Built target zed_wrapper_generate_messages_py
[ 76%] Built target zed_wrapper_generate_messages_eus
[ 74%] Built target zed_wrapper_generate_messages_lisp
[ 91%] Built target zed_wrapper_generate_messages_cpp
[ 91%] Built target zed_wrapper_generate_messages
make[2]: *** No rule to make target '/usr/local/cuda-10.2/lib64/libcublas.so', needed by '/home/scarf/zed_ws/devel/lib/libZEDWrapper.so'. Stop.
CMakeFiles/Makefile2:2537: recipe for target 'zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/all' failed
make[1]: *** [zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

Myzhar, 2020-01-15:
Hi @Scarf195 
Can you verify that `/usr/local/cuda-10.2/lib64/libcublas.so` is present in your CUDA 10.2 installation?

PS Italian is not a problem for me ;-)

Scarf195, 2020-01-15:
scarf@ScarPC:~$ /usr/local/cuda-10.2/lib64/libcublas.so
bash: /usr/local/cuda-10.2/lib64/libcublas.so: No such file or directory

Scarf195, 2020-01-15:
Sure I did something wrong with the CUDA installation, but I can't understand what, I followed the deb (local) installation type.

Myzhar, 2020-01-15:
Ok, I found this thread on the Nvidia forum discussing your problem for CUDA 10.1:
https://devtalk.nvidia.com/default/topic/1047981/cuda-setup-and-installation/cublas-for-10-1-is-missing/1

it seems that the cublas library has changed position starting from CUDA 10.1.

If you do not require CUDA 10.2 I suggest you to install CUDA 10 and the relative ZED SDK v2.8.5.

Otherwise you can follow the suggestions in that thread forum while we search for a direct solution.

Scarf195, 2020-01-16:
Hi, I've installed CUDA 10 and its relative ZED SDK v2.8.5 and finally I did the catkin_make.
Unfortunately when I type roslaunch zed_wrapper zed.launch I receive this error:

[ERROR] [1579175887.327683842]: Failed to load nodelet [/zed/zed_node] of type [zed_wrapper/ZEDWrapperNodelet] even after refreshing the cache: Failed to load library /home/scarpc/zed_ws/devel/lib//libZEDWrapper.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libnppc.so.10.0: cannot open shared object file: No such file or directory)
[ERROR] [1579175887.327801928]: The error before refreshing the cache was: Failed to load library /home/scarpc/zed_ws/devel/lib//libZEDWrapper.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libnppc.so.10.0: cannot open shared object file: No such file or directory)

Do you know what is this error?

Scarf195, 2020-01-16:
Just done but the problem still not fixed

Myzhar, 2020-01-16:
Even closing and reopening the console to reload the `.bashrc` file?
You can force it manually:
`source ~/catkin_ws/devel/setup.bash`

Scarf195, 2020-01-16:
Yes, same error.
I tried to do first  "sudo apt get-update" and after "sudo apt-get dist-upgrade" with this error:
Lettura elenco dei pacchetti... Fatto
Generazione albero delle dipendenze       
Lettura informazioni sullo stato... Fatto
È utile eseguire "apt --fix-broken install" per correggere ciò.
I seguenti pacchetti hanno dipendenze non soddisfatte:
 libnvidia-ifr1-435 : Dipende: libnvidia-gl-435 ma non è installato
 libnvidia-ifr1-435:i386 : Dipende: libnvidia-gl-435:i386 ma non è installato
 nvidia-driver-435 : Dipende: libnvidia-gl-435 (= 435.21-0ubuntu0.18.04.2) ma non è installato
                     Raccomanda: libnvidia-gl-435:i386 (= 435.21-0ubuntu0.18.04.2)
E: Dipendenze non soddisfatte. Provare "apt --fix-broken install" senza pacchetti (o specificare una soluzione).

Sorry again for the italian version.

Scarf195, 2020-01-16:
> Yes, same error.
> I tried to do first "sudo apt get-update" and after "sudo apt-get dist-upgrade" with this error:
> Lettura elenco dei pacchetti... Fatto
> Generazione albero delle dipendenze
> Lettura [...]

Fixed this error, but same error in roslaunch...

Myzhar, 2020-01-21:
There is a new installer online for the SDK v2.8.5 with CUDA 10.2 support that fixes the issue with `libcublas`:
https://www.stereolabs.com/developers/release/

closed 2020-01-21 by Myzhar

Scarf195, 2020-01-22:
Hi, thanks for the info, I've reinstalled SDK v2.8.5 with his CUDA 10.2 but unfortunately the error persists, but now I found a solution in this topic:
https://gist.github.com/glhr/5622057692e4cfd2cae8302e6eb3d502
Hope it is useful :)