stereolabs/zed-ros-wrapper #507: Multiple Zed Network

bsubbaraman, 2020-01-20:
Hi all,

I'm having trouble trying to launch multiple zed cameras, each attached to a different computer. I launch each one in it's own namespace and using a different map frame id, using the usual zed.launch command. However, when I look at the tf_tree, there is only one instance of the odom to base link to zed left/right camera frames at any one point, and it bounces from camera to camera.

Is there anything else that needs to be done to launch three zeds simultaneously? Do I need to change all the names in the common.yaml and urdf files to be distinct? Any direction is greatly appreciated. I'm using Zed SDK 2.8 on Ubuntu 16.04.

Thank you!!

Myzhar, 2020-01-20:
Hi @bsubbaraman 
You should use a different name for the root link of each ZED cameras.
Each node should load a different `common.yaml` file and a different `zed.urdf` where the parameter `camera_frame` has a different value.
For example you should use `zed1_camera_center` for the first camera, `zed2_camera_center` for the second camera and so on... replacing the link name `zed_camera_center` for the relative URDF file

bsubbaraman, 2020-01-21:
Ah, thank you! Tried this just as you responded, but it's nice to know this is the correct way to do things.

Thanks for your always prompt responses, @Myzhar!

closed 2020-01-21 by bsubbaraman