stereolabs/zed-ros-wrapper #508: Crash jetson tx2

h-ss, 2020-01-22:
Hello,

I'm using the stereo zed with ros. I have an abnormal behavior, I launched the zed wrapper on the jetson tx2 and rviz on my laptop. At first it was all good, but lately the tx2 crashed (like a shutdown). So I tried to launch the zed wrapper on tx2 and on my laptop I tried "rostopic echo /zed_node/rgb/image_rect_color" and shorlty after the tx2 crashes. Instead if I try "rostopic echo /zed_node/rgb/image_rect_color/compressed" it lasts much longer.

What could be the cause, why does the tx2 go directly in crash?

Myzhar, 2020-01-22:
Hi @h-ss 
according to what you are describing I can say that it is not a software problem, but an hardware problem.
Usually when a TX2 shuts down the causes are normally two:
1) Excessive heat badly dissipated (verify that the fan is running at amximum speed)
2) Source power is not enough. Are you using the default power source with the development kit?

h-ss, 2020-01-23:
I'm not sure about the hardware problem because I used the tx2 with lidar for hours without problems, but when I try with the zed the problem occurs.
I try an end to end connection with LAN cable beetween tx2 and my laptop, it works well..
I don't understand why, thank you for your help

Myzhar, 2020-01-23:
Ok, so you are facing the same error faced by another user.
The TX2 is not frozen, but it is the WIFI bandwidth that is saturated and on the remote machine you cannot receive any data causing the SSH connection to freeze.

You can decrease the required bandwidth in many ways:
- reduce FPS
- reduce resolution
- if using pointcloud reduce the publishing frequency

Myzhar, 2020-02-18:
@h-ss is this issue solved? Please close it if you do not need further help

closed 2020-02-20 by h-ss