stereolabs/zed-ros-wrapper #510: Where can i find what gpu is selected when running

Codeguyross, 2020-01-29:
Sorry for the question if I am in the wrong place. I am very new to both zed and ros. I am trying to figure out how to know exactly what GPU is selected by "gpu_id = -1" in camera config file. If it helps, here is what I see when I launch an example below. Note the line that reads "/zed/zed_node/general/gpu_id: -1". What are my options? is -1 the only option? 

```
SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /zed/zed_description: <?xml version="1....
 * /zed/zed_node/auto_exposure_gain: True
 * /zed/zed_node/auto_whitebalance: True
 * /zed/zed_node/brightness: 4
 * /zed/zed_node/confidence: 97
 * /zed/zed_node/contrast: 4
 * /zed/zed_node/depth/depth_stabilization: 1
 * /zed/zed_node/depth/max_depth: 10.0
 * /zed/zed_node/depth/min_depth: 0.7
 * /zed/zed_node/depth/openni_depth_mode: 0
 * /zed/zed_node/depth/quality: 4
 * /zed/zed_node/depth/sensing_mode: 0
 * /zed/zed_node/exposure: 100
 * /zed/zed_node/gain: 100
 * /zed/zed_node/general/base_frame: base_link
 * /zed/zed_node/general/camera_flip: False
 * /zed/zed_node/general/camera_frame: zed_camera_center
 * /zed/zed_node/general/camera_model: zed
 * /zed/zed_node/general/frame_rate: 15
 * /zed/zed_node/general/gpu_id: -1
 * /zed/zed_node/general/left_camera_frame: zed_left_camera_f...
 * /zed/zed_node/general/left_camera_optical_frame: zed_left_camera_o...
 * /zed/zed_node/general/resolution: 1
 * /zed/zed_node/general/right_camera_frame: zed_right_camera_...
 * /zed/zed_node/general/right_camera_optical_frame: zed_right_camera_...
 * /zed/zed_node/general/self_calib: True
 * /zed/zed_node/general/serial_number: 0
 * /zed/zed_node/general/svo_compression: 2
 * /zed/zed_node/general/verbose: True
 * /zed/zed_node/general/zed_id: 0
 * /zed/zed_node/hue: 0
 * /zed/zed_node/mapping/fused_pointcloud_freq: 1.0
 * /zed/zed_node/mapping/mapping_enabled: False
 * /zed/zed_node/mapping/resolution: 0
 * /zed/zed_node/mat_resize_factor: 1.0
 * /zed/zed_node/point_cloud_freq: 5.0
 * /zed/zed_node/pos_tracking/fixed_cov_value: 1e-6
 * /zed/zed_node/pos_tracking/fixed_covariance: False
 * /zed/zed_node/pos_tracking/fixed_z_value: 0.0
 * /zed/zed_node/pos_tracking/floor_alignment: False
 * /zed/zed_node/pos_tracking/init_odom_with_first_valid_pose: True
 * /zed/zed_node/pos_tracking/initial_base_pose: [0.0, 0.0, 0.0, 0...
 * /zed/zed_node/pos_tracking/map_frame: map
 * /zed/zed_node/pos_tracking/odometry_db: 
 * /zed/zed_node/pos_tracking/odometry_frame: odom
 * /zed/zed_node/pos_tracking/path_max_count: -1
 * /zed/zed_node/pos_tracking/path_pub_rate: 2.0
 * /zed/zed_node/pos_tracking/publish_map_tf: False
 * /zed/zed_node/pos_tracking/publish_pose_covariance: True
 * /zed/zed_node/pos_tracking/publish_tf: True
 * /zed/zed_node/pos_tracking/spatial_memory: True
 * /zed/zed_node/pos_tracking/two_d_mode: False
 * /zed/zed_node/pos_tracking/world_frame: map
 * /zed/zed_node/saturation: 4
 * /zed/zed_node/sharpness: 3
 * /zed/zed_node/stream: 
 * /zed/zed_node/svo_file: 
 * /zed/zed_node/video/color_enhancement: True
 * /zed/zed_node/whitebalance_temperature: 4200

NODES
[... 111 lines cut ...]
[ INFO] [1580333555.753553979]: Advertised on topic /zed/zed_node/confidence/confidence_map
[ INFO] [1580333555.753905871]: Advertised on topic /zed/zed_node/disparity/disparity_image
[ INFO] [1580333555.754240496]: Advertised on topic /zed/zed_node/point_cloud/cloud_registered
[ INFO] [1580333555.754589195]: Advertised on topic /zed/zed_node/pose
[ INFO] [1580333555.754932857]: Advertised on topic /zed/zed_node/pose_with_covariance
[ INFO] [1580333555.755268154]: Advertised on topic /zed/zed_node/odom
[ INFO] [1580333555.755594333]: Advertised on topic /zed/zed_node/path_odom
[ INFO] [1580333555.755902824]: Advertised on topic /zed/zed_node/path_map
[ INFO] [1580333555.775853534]: Static transform Sensor to Base [zed_left_camera_frame -> base_link]
[ INFO] [1580333555.775882272]:  * Translation: {0.000,-0.060,0.000}
[ INFO] [1580333555.775890372]:  * Rotation: {0.000,-0.000,0.000}
[ INFO] [1580333555.775911897]: Static transform Sensor to Camera Center [zed_left_camera_frame -> zed_camera_center]
[ INFO] [1580333555.775921662]:  * Translation: {0.000,-0.060,0.000}
[ INFO] [1580333555.775934460]:  * Rotation: {0.000,-0.000,0.000}
[ INFO] [1580333555.775952673]: Static transform Camera Center to Base [zed_camera_center -> base_link]
[ INFO] [1580333555.775964490]:  * Translation: {0.000,0.000,0.000}
[ INFO] [1580333555.775975874]:  * Rotation: {0.000,-0.000,0.000}
[ INFO] [1580333555.893028222]: *** Starting Positional Tracking ***
[ INFO] [1580333555.893064919]:  * Waiting for valid static transformations...
[ INFO] [1580333555.993258393]: Initial ZED left camera pose (ZED pos. tracking): 
[ INFO] [1580333555.993410332]:  * T: [0,0.06,0]
[ INFO] [1580333555.993566880]:  * Q: [0,0,0,1]
[ INFO] [1580333556.042223145]: Odometry aligned to last tracking pose

```

Myzhar, 2020-01-30:
Hi @Codeguyross 
to be sure to use the GPU that you want you can set its GPUid changing the `general/gpu_id` param in [`common.yaml`](https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/params/common.yaml)

If you use the value `-1` the ZED SDK will choose the better GPU for you and you can verify on which GPU the zed_wrapper is running using the `nvidia_smi` CLI command.

Furthermore we will add this information in the ZED Diagnostic to give a more complete ZED information.

Can I ask you why you need this information?

Codeguyross, 2020-01-31:
i was looking to confirm if the zed was actually using the gpu or had failed to use the gpu and was instead using the cpu. Because I wasn't sure what integer i could set the gpu_id, and I couldn't see the gpu the zed was using, I couldnt be 100% that the GPU portion was working. 

Is there documentation somewhere listing available options beyond the default value for parameters in the common.yaml file? The  root of this question is coming from having poor results when trying to create a scan of an area using the zed and so I would like more options with parameters to be able to try to correct the poor quality scans.

Myzhar, 2020-01-31:
This is the right place for this kind of question and you're always welcome.
About the use of the GPU, the ZED SDK needs CUDA and the depth processing (and many other algorithms) runs only on the GPU.
About the documentation, the right place is the Stereolabs website: 
https://www.stereolabs.com/docs/ros/zed_node/

Do not hesitate to open new issues if you need more support.

closed 2020-02-08 by Myzhar