stereolabs/zed-ros-wrapper #513: Object Avoidance using ZED camera

Scarf195, 2020-02-04:
Hi, I would like to use ZED-ROS-Wrapper to get an object avoidance using a rover with a Jetson Nano and a ZED Camera. My idea is to send via ROS a warning when an obstacle is near the rover. Do you have any suggestion to do this? I have to use rtabmap to do this task? I'm new in ROS and and I don't know where to start...

Myzhar, 2020-02-04:
Hi @Scarf195 
RTABmap is used to create a map. To perform obstacle avoidance you have to possible roads:
1) write your own new node that subscribes to point cloud or depth map topics and analyze them to extract obstacles
2) use the "navigation stack" that is used to create a map, localize the robot in the map, extract dynamic obstacles and avoid them

More info about 2 are available here:
http://wiki.ros.org/navigation
http://docs.ros.org/melodic/api/robot_localization/html/index.html

Scarf195, 2020-02-05:
Hi @Myzhar, thank you for your fast reply.
About RTABmap there is the possibility to observe obstacle subscribing to point cloud, as explained here http://wiki.ros.org/rtabmap_ros#rtabmap_ros.2BAC8-obstacles_detection, that's why I told you about that.
Do you think this could be useful?

Myzhar, 2020-02-05:
Sure. That feature can be very useful because you will get the pointcloud relative only to the obstacles filtering everything else.

Scarf195, 2020-02-05:
And in your zed_rtabmap_example is this also implemented?

Myzhar, 2020-02-05:
The `zed_rtabmap_example` runs an instance of RTABmap with all its features enabled.
The example starts a pre-configured RVIZ instance, in the left menu you can find `Obstacles`.
If you check it you can see how RTABmap publishes the point cloud of the detected obstacles.

[image]

Scarf195, 2020-02-14:
Hi @Myzhar, thank you very much for your reply, I've successfully implemented Navigation Stack, getting out a cmd_vel which I have to control my rover and for this I want to exploit the odometry get from the ZED, but it publishes only information about the position.
Now my question is: is it possible to get directly the velocity by ZED camera from his position?

Myzhar, 2020-02-14:
Hi @Scarf195 this is a feature that we plan to add in one of the next release.
Publishing the twist vector requires a few coordinate transformations from the left camera frame to the odometry frame and we must be sure that everything is correct before releasing.

Scarf195, 2020-03-27:
Hi @Myzhar, thanks for your aid, I've succesfully implemented autonomous navigation using rtabmap_ros for SLAM approach and Navigation Stack.

I'm worried about GPU usage that quickly increase during execution, especially when I use rviz from remote PC to visualize the map and the 3D point cloud and right now I'm changing the rtabmap.yaml and zed common.yaml trying to reducing that usage but without success.
Is this problem due to the fact that I'm using a Jetson Nano or maybe I have to change something in the file?

This is the zed common.yaml:
[common _.txt](https://github.com/stereolabs/zed-ros-wrapper/files/4394684/common._.txt)

and this is the rtabmap.yaml:
[rtabmap_.txt](https://github.com/stereolabs/zed-ros-wrapper/files/4394689/rtabmap_.txt)

For navigation Stack I'm using the zed_nodelet_example for publishing in zed/scan and  zed/point_cloud_registered, while for rtabmap I'm using only depth and rgb image.

Myzhar, 2020-03-27:
If you use Rviz on the Jetson Nano it is normal that the requested computational power is very high.
Rviz is an heavy application also for a laptop.

Scarf195, 2020-03-27:
No sorry, I said that badly, I'm not using RViz on Jetson Nano but on my laptop that receive in streaming data from Jetson.

Myzhar, 2020-03-27:
Ah ok, in that case it does not affect the Jetson Nano, but it depends on how many topics you subscribe from the remote PC.
If roscore is running on the Nano (that is the best thing because it can continue to work even if the network communication drops), a the more the subscribed topics requires, the more computational power is required... serialization requires memcpy...

tensarflow, 2020-03-30:
@Scarf195 Hey! I am going to implement a SLAM+Obstacle Avoidance system in my AGV and I'm still not sure what configuration to use. Would you recommend using a ZED 2 camera? I'm kind of sensing that there is a little more community around the Intel RealSense products? (Because its cheaper I suppose)
On the other hand I'd have to use either the D435i with an IMU inside, to do the Mapping, SLAM and obstacle avoidance on my host (, which is kind of a bummer for me because it would affect latency etc), or I'd have to use the T265 for SLAM (250Hz) and only do the mapping and obstacle avoidance with a D435 on my host. But honestly why would I use two seperate devices for something I can do with just one (like ZED 2 or D435i).
So I just wanted to ask if you are satisfied with the ZED 2? How are the latencies? And one explicit question: Does the ZED 2 camera provide you with any other data then the pointcloud? Is the ZED calculating the odometry or your host? Thanks in advance :)

Myzhar, 2020-03-30:
Hi @tensarflow 
all the ZED cameras provide more than the simple pointcloud.
In particular they provide odometry, pose fixed with relocation, mapping and other useful data.
The ZED2 is the most advanced model, with improved optics and a full stack of sensors adding barometer and magnetometer to the "simple" IMU.
Differently from the Realsense cameras all the algorithms run on the host, not on the device, this allows to provide more advanced features in a scalable way.

tensarflow, 2020-03-30:
Thanks for the quick response.
So is it possible to use only the ZED2 to do SLAM and dynamic obstacle avoidance? And which part of that task would be done on the host and which on the ZED?

Myzhar, 2020-03-30:
Sure, you can perform both tasks.
On the ZED is performed only frame synchronization and sensor data acquisition.
The ZED SDK can perform SLAM internally and provide you with 3D map and position.
You must implement the navigation and obstacle detection algorithms.

tensarflow, 2020-03-30:
Thanks!
And how are the framerates being handled? I have seen a few ZED-SLAM examples online. They are pretty accurate but also really slow. Is it for example possible to turn off the simultaneous mapping after I captured the whole area, to reduce latency? What would the framerate be then? And would it in that scenario still be able to relocate itself in that map?

Myzhar, 2020-03-30:
You set the framerate of grabbing and you tune all the parameters to reach the mapping rate that you need. You surely can disable mapping when you think that the map is completed.
Relocation is still active even if mapping is disabled.
The final framerate depends on the computational power of the host, remember that you need a CUDA enabled host: a laptop/desktop with Nvidia GPU card or an embedded Nvidia Jetson device.

Myzhar, 2020-03-30:
For more information I suggest you to look at the online ZED developer guide, all these problematics are discussed in details:
https://www.stereolabs.com/docs/

tensarflow, 2020-03-30:
Thanks for the link. I've read about the relocalization. I have an Nvidia Jetson device, but according to the docs relocalization happens in the camera right? So it would then depend on the selected video format, correct? (Supposed I don't do any highlevel computation on the host)

Myzhar, 2020-03-30:
This depends on how you design your algorithm

closed 2022-04-19 by github-actions[bot]