stereolabs/zed-ros-wrapper #514: Unable to complie Zed_SDK with Nvidia GTX 1060 Cuda 10.2

q8wwe, 2020-02-04:
My PC is Asus with GTX 1060 I updated the driver to 440.33 with cuda 10.2.

First, when I install Zed SDK version 3.01 which compatible with cuda 10.2 it gives me the following warning :
E: Unable to locate package libturbojpeg0-dev
[image]
-----------------------------------------------
secondly, when I try to comply (catkin_make -DCMAKE_BUILD_TYPE=Release) it will fail:-

> make[2]: *** No rule to make target 'true', needed by '/home/moh/catkin_wss/devel/lib/libZEDWrapper.so'.  Stop.
CMakeFiles/Makefile2:3669: recipe for target 'zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/all' failed
make[1]: *** [zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed

kindly advise how to solve these problems.
Best Regards

Myzhar, 2020-02-05:
Hi @q8wwe 
have you deleted the folders `devel` and `build` in your catkin workspace after updating to SDK v3.0.1?
Deleting those folders clears the CMake cache and can solve the compiling problem

q8wwe, 2020-02-05:
I did that, delete build, devil folder in catkin and compile but unfortunately the same problems.

Myzhar, 2020-02-05:
About the first problem, have you tried to manually install `libturbojpeg0-dev`?
```
$ sudo apt update
$ sudo apt install libturbojpeg0-dev
```

q8wwe, 2020-02-05:
I tried to install libturbojpeg0-dev but it didn't work in Ubuntu 16.04 .
when I run: sudo apt install libturbojpeg0-dev
I got : E: Unable to locate package libturbojpeg0-dev

Myzhar, 2020-02-05:
@q8wwe you are speaking about Ubuntu 16.04. 

The installer for Ubuntu 16.04 does not require `libturbojpeg0-dev`, maybe you downloaded the version of the SDK for Ubuntu 18.04.

Can you download again the version for Ubuntu 16.04 with CUDA 10.2 support?

q8wwe, 2020-02-05:
currently, I'm using kernel 4.15.0-62-generic shall I downgrade to 4.13.X ?

Myzhar, 2020-02-05:
I guess it's not a kernel problem, but a wrong SDK version. Can you double check the name of the file you downloaded from Stereolabs' website?

q8wwe, 2020-02-05:
ZED SDK name : ZED_SDK_Ubuntu16_cuda10.2_v3.0.1.run

Myzhar, 2020-02-05:
At this point I cannot exclude that it is a kernel issue. 
The official Kernel version for Ubuntu 16.04.x is v4.4.x.

q8wwe, 2020-02-05:
I tried different nvidia driver 410,418,415,430,440 ,I tried with Cuda 10.2 , cuda 10.1 , cuda 10.
but I always run on the previous problems.

is there any possible solution ?

Myzhar, 2020-02-05:
We are investigating. You are the first with a similar issue.

q8wwe, 2020-02-06:
I made a fresh install to ubuntu 16.04 then follow the below steps:
installed Cuda 10.2
installed ZED_SDK_Ubuntu16_cuda10.2_v3.0.1.run
download and compile Ros_Wrapper from https://github.com/stereolabs/zed-ros-wrapper/tree/master/zed_wrapper

but the problems still exist.

Mr. Myzhar , if you need any information from me please let me know.
I'm working on Asus with GTX 1060 GPU.

q8wwe, 2020-02-06:
I downgraded to cuda 9 and ZDK 2.8.5 and use ros _wrapper 2.8.x

`:~/catkin_ws$ catkin_make
Base path: /home/moh/catkin_ws
Source space: /home/moh/catkin_ws/src
Build space: /home/moh/catkin_ws/build
Devel space: /home/moh/catkin_ws/devel
Install space: /home/moh/catkin_ws/install
####
#### Running command: "cmake /home/moh/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/moh/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/moh/catkin_ws/install -G Unix Makefiles" in "/home/moh/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/moh/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/kinetic
-- This workspace overlays: /opt/ros/kinetic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.12", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/moh/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.12") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.20
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 9 packages in topological order:
-- ~~  - zed_ar_track_alvar_example
-- ~~  - zed_depth_sub_tutorial
-- ~~  - zed_display_rviz
-- ~~  - zed_nodelet_example
-- ~~  - zed_ros (metapackage)
-- ~~  - zed_rtabmap_example
-- ~~  - zed_tracking_sub_tutorial
-- ~~  - zed_video_sub_tutorial
-- ~~  - zed_wrapper
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'zed_ar_track_alvar_example'
-- ==> add_subdirectory(zed-ros-wrapper/examples/zed_ar_track_alvar_example)
-- +++ processing catkin package: 'zed_depth_sub_tutorial'
-- ==> add_subdirectory(zed-ros-wrapper/tutorials/zed_depth_sub_tutorial)
-- +++ processing catkin package: 'zed_display_rviz'
-- ==> add_subdirectory(zed-ros-wrapper/zed_display_rviz)
-- +++ processing catkin package: 'zed_nodelet_example'
-- ==> add_subdirectory(zed-ros-wrapper/examples/zed_nodelet_example)
-- +++ processing catkin metapackage: 'zed_ros'
-- ==> add_subdirectory(zed-ros-wrapper/zed_ros)
-- +++ processing catkin package: 'zed_rtabmap_example'
-- ==> add_subdirectory(zed-ros-wrapper/examples/zed_rtabmap_example)
-- +++ processing catkin package: 'zed_tracking_sub_tutorial'
-- ==> add_subdirectory(zed-ros-wrapper/tutorials/zed_tracking_sub_tutorial)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- +++ processing catkin package: 'zed_video_sub_tutorial'
-- ==> add_subdirectory(zed-ros-wrapper/tutorials/zed_video_sub_tutorial)
-- +++ processing catkin package: 'zed_wrapper'
-- ==> add_subdirectory(zed-ros-wrapper/zed_wrapper)
-- A library with BLAS API found.
-- Found CUDA: /usr/local/cuda-9.0 (found suitable version "9.0", minimum required is "9.0") 
-- Found CUDA: /usr/local/cuda-9.0 (found version "9.0") 
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- zed_wrapper: 0 messages, 9 services
-- Configuring done
-- Generating done
-- Build files have been written to: /home/moh/catkin_ws/build
####
#### Running command: "make -j12 -l12" in "/home/moh/catkin_ws/build"
####
[  3%] Built target zed_depth_sub
[  6%] Built target zed_video_sub
[  9%] Built target sub
[ 13%] Built target zed_tracking_sub
[ 14%] Generating dynamic reconfigure files from cfg/Zed.cfg: /home/moh/catkin_ws/devel/include/zed_wrapper/ZedConfig.h /home/moh/catkin_ws/devel/lib/python2.7/dist-packages/zed_wrapper/cfg/ZedConfig.py
[ 14%] Built target _zed_wrapper_generate_messages_check_deps_toggle_led
[ 14%] Built target _zed_wrapper_generate_messages_check_deps_reset_tracking
[ 14%] Built target _zed_wrapper_generate_messages_check_deps_start_remote_stream
/home/moh/catkin_ws/build/zed-ros-wrapper/zed_wrapper/setup_custom_pythonpath.sh: 5: exec: /home/moh/catkin_ws/src/zed-ros-wrapper/zed_wrapper/cfg/Zed.cfg: Permission denied
zed-ros-wrapper/zed_wrapper/CMakeFiles/zed_wrapper_gencfg.dir/build.make:63: recipe for target '/home/moh/catkin_ws/devel/include/zed_wrapper/ZedConfig.h' failed
make[2]: *** [/home/moh/catkin_ws/devel/include/zed_wrapper/ZedConfig.h] Error 126
CMakeFiles/Makefile2:2235: recipe for target 'zed-ros-wrapper/zed_wrapper/CMakeFiles/zed_wrapper_gencfg.dir/all' failed
make[1]: *** [zed-ros-wrapper/zed_wrapper/CMakeFiles/zed_wrapper_gencfg.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 14%] Built target _zed_wrapper_generate_messages_check_deps_reset_odometry
[ 14%] Built target _zed_wrapper_generate_messages_check_deps_stop_remote_stream
[ 14%] Built target _zed_wrapper_generate_messages_check_deps_start_svo_recording
[ 14%] Built target _zed_wrapper_generate_messages_check_deps_set_led_status
[ 14%] Built target _zed_wrapper_generate_messages_check_deps_set_pose
[ 14%] Built target _zed_wrapper_generate_messages_check_deps_stop_svo_recording
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed
`

still can't compile ........

Myzhar, 2020-02-06:
It's not a problem with the wrapper, but with your ROS installation. 
`[ 14%] Built target _zed_wrapper_generate_messages_check_deps_start_remote_stream
/home/moh/catkin_ws/build/zed-ros-wrapper/zed_wrapper/setup_custom_pythonpath.sh: 5: exec: /home/moh/catkin_ws/src/zed-ros-wrapper/zed_wrapper/cfg/Zed.cfg: Permission denied`
This is a permissions problem, you used a `sudo` somewhere when it was not required

q8wwe, 2020-02-06:
how can I fix this -> remove the sudo .
 if I recreate catkin workspace would it work !!

I'm sorry for any inconvenient

q8wwe, 2020-02-07:
downgraded my Nvidia card to 384 , install cuda 9.0 , install sdk 2.8.5  and downloaded ros_wrapper 2.8.x.
that's configuration worked for me.

I have question about ZED SDK and Gazebo:
can I use ZED  sdk to produce Depth image with gazebo simulation, I need to work with zed_nodelet_laserscan.launch

Myzhar, 2020-02-09:
@q8wwe the ZED SDK does not accept external input if not an SVO file or a physical Stereolabs camera. For this reason you cannot directly use it in Gazebo

closed 2020-02-18 by Myzhar