stereolabs/zed-ros-wrapper #515: CUDE_nppi_LIBRARY set to NOTFOUND compiling zed_wrapper_node

UltharPeliki7, 2020-02-05:
Specific error message received was the following:

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CUDA_nppi_LIBRARY (ADVANCED)
    linked by target "ZEDWrapper" in directory /home/monarch/catkin_ws/src/zed-ros-wrapper/zed_wrapper
    linked by target "ZEDWrapper" in directory /home/monarch/catkin_ws/src/zed-ros-wrapper/zed_wrapper
    linked by target "zed_wrapper_node" in directory /home/monarch/catkin_ws/src/zed-ros-wrapper/zed_wrapper
    linked by target "zed_wrapper_node" in directory /home/monarch/catkin_ws/src/zed-ros-wrapper/zed_wrapper

-- Configuring incomplete, errors occurred!
See also "/home/monarch/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/monarch/catkin_ws/build/CMakeFiles/CMakeError.log".

SDK Version is 3.0.1, CUDA 10, Nvidia gtx750ti GPU, as for wrapper version, I was following the stereolabs installation guide verbatim, I would assume the most recent revision. Any assistance in rectifying this would be great.

Myzhar, 2020-02-05:
Hi @UltharPeliki7 
have you download the correct version of the ZED SDK for CUDA 10.0?
In the download page you can find a version for CUDA 10.0 and a version for CUDA 10.2.

Another cause of this kind of problems is a library update. Try to delete the folders `devel` and `build` in your catkin workspace and to compile again.

UltharPeliki7, 2020-02-06:
Yes, I am using cuda_10.0.130_410.48_linux.run and ZED_SDK_Ubuntu16_cuda10.0_v3.0.1.run. I have also tried deleting devel and build in my catkin workspace, to no avail. Last night, I went on a quest and removed all Nvidia related things on my install, which broke my DE. I am now running the Driver Version: 430.64       CUDA Version: 10.1 according to nvidia-smi. When I uninstall ver 430 and use the driver that comes with the CUDA downloader referenced above, I am not able to log into my DE, I can only access my CLI. I have not attempted a lower Nvidia driver version.

UltharPeliki7, 2020-02-06:
It appears the solution is to use the following from https://stackoverflow.com/questions/46584000/cmake-error-variables-are-set-to-notfound: Although it was intended for an opencv error, it fixes this missing library issue. That being said, I still am not able to compile. I will post that in the next comment for sake of clarity.

Change in FindCUDA.cmake the nppi library to the several splitted ones. This has to be done in 3 places. Remember this change is just to make it work with CUDA 9.0, I am not doing checks for version or anything, which should be done if you plan to give it to different people with different CUDA versions.

1) look for the line with:

find_cuda_helper_libs(nppi)

and replace it with the lines:

  find_cuda_helper_libs(nppial)
  find_cuda_helper_libs(nppicc)
  find_cuda_helper_libs(nppicom)
  find_cuda_helper_libs(nppidei)
  find_cuda_helper_libs(nppif)
  find_cuda_helper_libs(nppig)
  find_cuda_helper_libs(nppim)
  find_cuda_helper_libs(nppist)
  find_cuda_helper_libs(nppisu)
  find_cuda_helper_libs(nppitc)

2) find the line:

set(CUDA_npp_LIBRARY "${CUDA_nppc_LIBRARY};${CUDA_nppi_LIBRARY};${CUDA_npps_LIBRARY}")

and change it to

set(CUDA_npp_LIBRARY "${CUDA_nppc_LIBRARY};${CUDA_nppial_LIBRARY};${CUDA_nppicc_LIBRARY};${CUDA_nppicom_LIBRARY};${CUDA_nppidei_LIBRARY};${CUDA_nppif_LIBRARY};${CUDA_nppig_LIBRARY};${CUDA_nppim_LIBRARY};${CUDA_nppist_LIBRARY};${CUDA_nppisu_LIBRARY};${CUDA_nppitc_LIBRARY};${CUDA_npps_LIBRARY}")

3) find the unset variables and add the new variables as well So, find:

unset(CUDA_nppi_LIBRARY CACHE)

and change it to:

unset(CUDA_nppial_LIBRARY CACHE)
unset(CUDA_nppicc_LIBRARY CACHE)
unset(CUDA_nppicom_LIBRARY CACHE)
unset(CUDA_nppidei_LIBRARY CACHE)
unset(CUDA_nppif_LIBRARY CACHE)
unset(CUDA_nppig_LIBRARY CACHE)
unset(CUDA_nppim_LIBRARY CACHE)
unset(CUDA_nppist_LIBRARY CACHE)
unset(CUDA_nppisu_LIBRARY CACHE)
unset(CUDA_nppitc_LIBRARY CACHE)

UltharPeliki7, 2020-02-06:
Linking CXX executable /home/monarch/catkin_ws/devel/lib/zed_wrapper/zed_wrapper_node
/opt/ros/kinetic/lib/libbondcpp.so: undefined reference to `ros::NodeHandle::createSteadyTimer(ros::WallDuration, boost::function<void (ros::SteadyTimerEvent const&)> const&, bool, bool) const'
/opt/ros/kinetic/lib/libbondcpp.so: undefined reference to `ros::SteadyTimer::setPeriod(ros::WallDuration const&, bool)'
/opt/ros/kinetic/lib/libbondcpp.so: undefined reference to `ros::SteadyTimer::~SteadyTimer()'
/opt/ros/kinetic/lib/libbondcpp.so: undefined reference to `ros::SteadyTimer::stop()'
collect2: error: ld returned 1 exit status
zed-ros-wrapper/zed_wrapper/CMakeFiles/zed_wrapper_node.dir/build.make:152: recipe for target '/home/monarch/catkin_ws/devel/lib/zed_wrapper/zed_wrapper_node' failed
make[2]: *** [/home/monarch/catkin_ws/devel/lib/zed_wrapper/zed_wrapper_node] Error 1
CMakeFiles/Makefile2:62021: recipe for target 'zed-ros-wrapper/zed_wrapper/CMakeFiles/zed_wrapper_node.dir/all' failed
make[1]: *** [zed-ros-wrapper/zed_wrapper/CMakeFiles/zed_wrapper_node.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

Myzhar, 2020-02-06:
This is a ROS library problem, is your environment correctly set?
If you are using `.bashrc` to automatically configure the environment verify that this command is present:
`source /opt/ros/kinetic/setup.bash`

UltharPeliki7, 2020-02-06:
Yes, at the bottom of the file: 
source /opt/ros/kinetic/setup.bash
source /home/monarch/catkin_ws/devel/setup.bash are present

uddhavbhattarai, 2020-02-06:
I am facing exactly same problem with ZED 2 camera.
CUDA 9.0
ZED SDK Version: 3.0.1

I already checked `bashrc` to configure the environment.

UltharPeliki7, 2020-02-07:
@uddhavbhattarai Which problem? The CUDA_nppi_library, or the next one? I outlined the solution for the change in FINDCuda which will fix the nppi library, as for the other problem, I still have not solved it. You can follow it at the following link: https://answers.ros.org/question/343455/libbondcppso-undefined-reference-to-core-ros-functions-packages-error/

airfield20, 2020-02-07:
I am also having this issue on the Zed 2 Camera with the CUDA_nppi_library. The cmake generation fails with 
```
-- Found CUDA: /usr/local/cuda (found suitable version "10.0", minimum required is "10") 
-- Found CUDA: /usr/local/cuda (found version "10.0") 
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- zed_wrapper: 2 messages, 13 services
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CUDA_nppi_LIBRARY (ADVANCED)
    linked by target "zed_wrapper_node" in directory /home/aaron/catkin_ws/src/zed-ros-wrapper/zed_wrapper
    linked by target "ZEDWrapper" in directory /home/aaron/catkin_ws/src/zed-ros-wrapper/zed_wrapper

-- Configuring incomplete, errors occurred!
See also "/home/aaron/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/aaron/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:2404: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

```
 I have the new sdk installed and have tested it with the sample projects that come with the sdk.

UltharPeliki7, 2020-02-08:
@airfield20 Please see my previous comment regarding the change needed to be made to FINDcuda. Let me know how it goes.

wangq0688, 2020-02-10:
> @airfield20 Please see my previous comment regarding the change needed to be made to FINDcuda. Let me know how it goes.

hi,I have the same problem. and i have changed FINDcuda.
error message still was

Please set them or make sure they are set and tested correctly in the CMake files:
CUDA_nppi_LIBRARY (ADVANCED)
    linked by target "ZEDWrapper" in directory /home/bot/catkin_ws/src/zed-ros-wrapper-master/zed_wrapper
    linked by target "ZEDWrapper" in directory /home/bot/catkin_ws/src/zed-ros-wrapper-master/zed_wrapper
    linked by target "zed_wrapper_node" in directory /home/bot/catkin_ws/src/zed-ros-wrapper-master/zed_wrapper
    linked by target "zed_wrapper_node" in directory /home/bot/catkin_ws/src/zed-ros-wrapper-master/zed_wrapper

Myzhar, 2020-02-10:
Hi @wangq0688 
are you using Ubuntu16 or Ubuntu18?

wangq0688, 2020-02-10:
> Hi @wangq0688 and @airfield20
> are you using Ubuntu16 or Ubuntu18?

ubuntu16

abhay1220, 2020-02-11:
@UltharPeliki7, I am still having the same issue even after making the changes in the FindCUDA.cmake file as you mentioned above. Kindly let me know if you know any solution for this.

Found CUDA: /usr/local/cuda (found suitable version "10.2", minimum required is "10.2") 
-- Found CUDA: /usr/local/cuda (found version "10.2") 
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- zed_wrapper: 2 messages, 13 services
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CUDA_nppi_LIBRARY (ADVANCED)
    linked by target "zed_wrapper_node" in directory /home/autosyslab/catkin_ws/src/zed-ros-wrapper/zed_wrapper
    linked by target "ZEDWrapper" in directory /home/autosyslab/catkin_ws/src/zed-ros-wrapper/zed_wrapper

-- Configuring incomplete, errors occurred!
See also "/home/autosyslab/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/autosyslab/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

UltharPeliki7, 2020-02-12:
@abhay1220 My fix is for ROS Kinetic, Ubuntu 16.04, CUDA 10, I am not sure it will hold for 10.2, and if you aren't also on kinetic or ubuntu 16.04, I don't think it will work. This is kind of blind leading the blind here, I'm still convinced all of this is magic voodoo at this point because sometimes it'll compile nicely, and othertimes it won't. One thing I have noticed is sometimes my catkin workspaces will overlay without me intentionally doing so, and that needs correcting.

Myzhar, 2020-02-12:
There was a dependency to CUDE_nppi_LIBRARY in CMakeFiles.txt that was not required anymore.
Can you pull the new version of the wrapper and try to compile it to confirm that the issue is fixed?

airfield20, 2020-02-13:
> @airfield20 Please see my previous comment regarding the change needed to be made to FINDcuda. Let me know how it goes.

The suggested workaround did not work

airfield20, 2020-02-13:
> Hi @wangq0688 and @airfield20
> are you using Ubuntu16 or Ubuntu18?

Ubuntu 16 with ros kinetic

Myzhar, 2020-02-13:
@airfield20 update to the new version. The issue should have been fixed

wangq0688, 2020-02-14:
I reinstalled the operating system and solved the problem.

closed 2020-02-14 by Myzhar