stereolabs/zed-ros-wrapper #519: unable to compile zed-ros-wrapper on jetson nano

Al-khattab, 2020-02-08:
I am trying to compile zed-wrapper on jetson nano, I'm following the instructions but I'm getting an error when I reach this line

`catkin_make -DCMAKE_BUILD_TYPE=Release
`

`[ 96%] Building CXX object zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/src/nodelet/src/zed_wrapper_nodelet.cpp.o
[ 96%] Building CXX object zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/src/tools/src/sl_tools.cpp.o
/home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp: In member function ‘bool zed_wrapper::ZEDWrapperNodelet::start_3d_mapping()’:
/home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:1160:52: error: invalid use of non-static data member ‘sl::SpatialMappingParameters::allowed_resolution’
     if(mMappingRes < sl::SpatialMappingParameters::allowed_resolution.first) {
                                                    ^~~~~~~~~~~~~~~~~~
In file included from /home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/tools/include/sl_tools.h:26:0,
                 from /home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/include/zed_wrapper_nodelet.hpp:28,
                 from /home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:21:
/usr/local/zed/include/sl/Camera.hpp:5475:75: note: declared here
         /*const*/ interval allowed_resolution = std::make_pair(0.01f, 0.2f);
                                                                           ^
/home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:1162:53: error: invalid use of non-static data member ‘sl::SpatialMappingParameters::allowed_resolution’
         mMappingRes = sl::SpatialMappingParameters::allowed_resolution.first;
                                                     ^~~~~~~~~~~~~~~~~~
In file included from /home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/tools/include/sl_tools.h:26:0,
                 from /home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/include/zed_wrapper_nodelet.hpp:28,
                 from /home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:21:
/usr/local/zed/include/sl/Camera.hpp:5475:75: note: declared here
         /*const*/ interval allowed_resolution = std::make_pair(0.01f, 0.2f);
                                                                           ^
/home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:1164:52: error: invalid use of non-static data member ‘sl::SpatialMappingParameters::allowed_resolution’
     if(mMappingRes > sl::SpatialMappingParameters::allowed_resolution.second) {
                                                    ^~~~~~~~~~~~~~~~~~
In file included from /home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/tools/include/sl_tools.h:26:0,
                 from /home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/include/zed_wrapper_nodelet.hpp:28,
                 from /home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:21:
/usr/local/zed/include/sl/Camera.hpp:5475:75: note: declared here
         /*const*/ interval allowed_resolution = std::make_pair(0.01f, 0.2f);
                                                                           ^
/home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:1166:53: error: invalid use of non-static data member ‘sl::SpatialMappingParameters::allowed_resolution’
         mMappingRes = sl::SpatialMappingParameters::allowed_resolution.second;
                                                     ^~~~~~~~~~~~~~~~~~
In file included from /home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/tools/include/sl_tools.h:26:0,
                 from /home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/include/zed_wrapper_nodelet.hpp:28,
                 from /home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:21:
/usr/local/zed/include/sl/Camera.hpp:5475:75: note: declared here
         /*const*/ interval allowed_resolution = std::make_pair(0.01f, 0.2f);
                                                                           ^
/home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:1174:64: error: invalid use of non-static data member ‘sl::SpatialMappingParameters::allowed_range’
     } else if(mMaxMappingRange < sl::SpatialMappingParameters::allowed_range.first) {
                                                                ^~~~~~~~~~~~~
In file included from /home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/tools/include/sl_tools.h:26:0,
                 from /home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/include/zed_wrapper_nodelet.hpp:28,
                 from /home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:21:
/usr/local/zed/include/sl/Camera.hpp:5492:68: note: declared here
         /*const*/ interval allowed_range = std::make_pair(2.f, 20.f);
                                                                    ^
/home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:1176:58: error: invalid use of non-static data member ‘sl::SpatialMappingParameters::allowed_range’
         mMaxMappingRange = sl::SpatialMappingParameters::allowed_range.first;
                                                          ^~~~~~~~~~~~~
In file included from /home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/tools/include/sl_tools.h:26:0,
                 from /home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/include/zed_wrapper_nodelet.hpp:28,
                 from /home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:21:
/usr/local/zed/include/sl/Camera.hpp:5492:68: note: declared here
         /*const*/ interval allowed_range = std::make_pair(2.f, 20.f);
                                                                    ^
/home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:1177:64: error: invalid use of non-static data member ‘sl::SpatialMappingParameters::allowed_range’
     } else if(mMaxMappingRange > sl::SpatialMappingParameters::allowed_range.second) {
                                                                ^~~~~~~~~~~~~
In file included from /home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/tools/include/sl_tools.h:26:0,
                 from /home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/include/zed_wrapper_nodelet.hpp:28,
                 from /home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:21:
/usr/local/zed/include/sl/Camera.hpp:5492:68: note: declared here
         /*const*/ interval allowed_range = std::make_pair(2.f, 20.f);
                                                                    ^
/home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:1179:58: error: invalid use of non-static data member ‘sl::SpatialMappingParameters::allowed_range’
         mMaxMappingRange = sl::SpatialMappingParameters::allowed_range.second;
                                                          ^~~~~~~~~~~~~
In file included from /home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/tools/include/sl_tools.h:26:0,
                 from /home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/include/zed_wrapper_nodelet.hpp:28,
                 from /home/ghattas/catkin_ws/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:21:
/usr/local/zed/include/sl/Camera.hpp:5492:68: note: declared here
         /*const*/ interval allowed_range = std::make_pair(2.f, 20.f);
                                                                    ^
zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/build.make:86: recipe for target 'zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/src/nodelet/src/zed_wrapper_nodelet.cpp.o' failed
make[2]: *** [zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/src/nodelet/src/zed_wrapper_nodelet.cpp.o] Error 1
CMakeFiles/Makefile2:681: recipe for target 'zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/all' failed
make[1]: *** [zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

Invoking "make -j4 -l4" failed
`

Myzhar, 2020-02-08:
Hi @Al-khattab 
Are you using the latest SDK v3.0?

Al-khattab, 2020-02-08:
Yes

Scarf195, 2020-02-08:
Hi, I got the same error also using jetson nano with Jetpack 4.3

Myzhar, 2020-02-08:
I check the wrapper and let you know when it's fixed. Sorry for the inconvenience

Myzhar, 2020-02-08:
You can pull the fixed ZED Wrapper from the `master` branch. Sorry again for the inconvenience it caused.

closed 2020-02-08 by Myzhar