stereolabs/zed-ros-wrapper #523: zed ROS wrapper not compiling with doxygen errors

thatrobotguy, 2020-02-13:
There are several errors popping up when trying to compile this ROS package with Ubuntu 1804 ROS melodic.

On branch `master` these errors are thrown:
```
- A library with BLAS API found.
-- Found CUDA: /usr/local/cuda-10.0 (found suitable version "10.0", minimum required is "10") 
-- Found CUDA: /usr/local/cuda-10.0 (found version "10.0") 
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- zed_wrapper: 2 messages, 13 services
-- Configuring done
CMake Error at zed-ros-wrapper/zed_wrapper/CMakeLists.txt:161 (add_custom_target):
  Cannot find source file:

    /home/devmachine2/ros1_ws/src/sick_scan/doxygen/.#html_step4.sh

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx

-- Generating done
-- Build files have been written to: /home/devmachine2/ros1_ws/build
Invoking "cmake" failed
```
On tag `v2.8.x`, these errors show up:
```
-- ==> add_subdirectory(zed-ros-wrapper/zed_wrapper)
-- A library with BLAS API found.
-- Found CUDA: /usr/local/cuda-10.0 (found suitable version "10.0", minimum required is "10") 
-- Found CUDA: /usr/local/cuda-10.0 (found version "10.0") 
-- Found OpenMP_C: -fopenmp  
-- Found OpenMP_CXX: -fopenmp  
-- Found OpenMP: TRUE   
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- zed_wrapper: 0 messages, 9 services
-- Configuring done
CMake Error at zed-ros-wrapper/examples/zed_ar_track_alvar_example/CMakeLists.txt:13 (add_custom_target):
  Cannot find source file:

    /home/devmachine2/ros1_ws/src/sick_scan/doxygen/.#html_step4.sh

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx

CMake Error at zed-ros-wrapper/zed_display_rviz/CMakeLists.txt:12 (add_custom_target):
  Cannot find source file:

    /home/devmachine2/ros1_ws/src/sick_scan/doxygen/.#html_step4.sh

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx

CMake Error at zed-ros-wrapper/examples/zed_nodelet_example/CMakeLists.txt:12 (add_custom_target):
  Cannot find source file:

    /home/devmachine2/ros1_ws/src/sick_scan/doxygen/.#html_step4.sh

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx

CMake Error at zed-ros-wrapper/examples/zed_rtabmap_example/CMakeLists.txt:12 (add_custom_target):
  Cannot find source file:

    /home/devmachine2/ros1_ws/src/sick_scan/doxygen/.#html_step4.sh

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx

CMake Error at zed-ros-wrapper/zed_wrapper/CMakeLists.txt:154 (add_custom_target):
  Cannot find source file:

    /home/devmachine2/ros1_ws/src/sick_scan/doxygen/.#html_step4.sh

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx

-- Generating done
-- Build files have been written to: /home/devmachine2/ros1_ws/build
Invoking "cmake" failed
```
Is there an extension related to `doxygen` that I am missing? 
I see that the errors point to a package called `sick_scan`, but that package compiles without a hitch when `zed-ros-wrapper` is not involved. I know the package `sick_scan` is not a problem, so why would `zed-ros-wrapper` cause doxygen errors? I am stumped as executable code has nothing to do with doxygen.
I would also like to mention that I am running the ZED SDK version `3.x.x`, so I am surprised to see that the branches and tags in this repo are not yet there.

Myzhar, 2020-02-13:
Hi @thatrobotguy ,
can you please try to compile the `zed-ros-wrapper` package in a clean catkin workspace?
Then you can add the other packages one by one to search for the one causing the conflict.

thatrobotguy, 2020-02-13:
Hey @Myzhar 
I moved all other packages outside of that directory and the package compiled. 
I am wondering then: Why are you doing a `GLOB` call from the `cmake` root? Wouldn't you only want to do that in the package?
I am thinking that this should be set by a `cmake` option rather than hard-code this behavior.
Maybe a switch for:
1. All off
2. Only this package
3. The entire workspace
That behavior makes more sense to me. What do you think?

Myzhar, 2020-02-13:
It is a feature used to import all the file of the package (and not only the source code) in the QtCreator IDE.
It never generated any issue. This is the first time with you

thatrobotguy, 2020-02-13:
Well, it is not a feature for me...
What should I replace those lines with then?
Should I change out
```
FILE(GLOB_RECURSE all_files ${CMAKE_SOURCE_DIR}/*)
add_custom_target(all_files_${PROJECT_NAME} SOURCES ${all_files})
```
to become this:
```
FILE(GLOB_RECURSE all_files ${CMAKE_CURRENT_LIST_DIR}/*)
add_custom_target(all_files_${PROJECT_NAME} SOURCES ${all_files})
```
or is there something else I need to do?

```

Myzhar, 2020-02-13:
You can simply remove them since they are not strictly necessary to compile the wrapper.

thatrobotguy, 2020-02-13:
Cool I will do that then. A simple switch could be added such that the user can select if they want that functionality or not using the `option` keyword in an `if()` statement.

closed 2020-02-13 by thatrobotguy